ramapcsx2/gbs-control #99: GBS Control and unsteady picture

rabidgerry, 2020-01-28:
When using gbs control the picture is set how I like it and saved on a preset.  However from time to time it will shake for a moment or have like a line go through the screen causing a distortion briefly.   Sometimes they can happen quite regularly and other time not so often, but it is really annoying and I'd like to fix it as it is distracting if playing a game for example.

I have only just started using this device with my Amiga and gbs - 8220 board.  Only clocked up a few hours with it so that is all the behaviour I have monitored so far.

I don't seem to suffer connections issues like others have noted from the heat sink interference (although no the gbs control cannot be located by any wifi device in my house since entering in my home wifi ssid and password) however I have asked about this in a separate Issue topic.

Mods I had done before the gbs control were removal of c11 and foil on the ram traces.  Not sure if this has any significance but just mentioning.

ramapcsx2, 2020-01-29:
I don't have enough information to identify the issue.  
Generally, you should try to leave the default presets as they are and correct the image position on your TV.  
If that's not possible, try to restrict image adjustments to the "move picture" controls. Do not use the scaling controls, if possible.

If you want to provide details, please enable the Development > Print Info feature and copy/paste a few lines here (5 lines or so is fine).  
I also need to know whether your Sync is single wire (CSync) or separate H+V Sync.
If using CSync, make sure the signal is properly attenuated with a ~470 Ohm series resistor before it goes into the GBS.

rabidgerry, 2020-01-29:
Ok before I continue, on my accord I tried figuring out a solution before you answered me.  What I initially tried worked briefly last night until the image became unstable and would also go completely black then back again, move around position for a bit before going black again.  This cycle did not seem as though it would end.

What I tried was the FrameTime Lock option because to me it described what I was experiencing in the couple of lines above it.

So today being another day, I switched it on, and basically have had it on for a few hours and the tear line does not happen anymore and the disruption I experienced last night (with unstable image and going blank then repeating)  did not happen either.  Is it fixed?  I do not know.  What is the likely hood that this might happen again?

To answer your questions,  I have one wire for sync and it has a 680ohm resistor between that rgb pin and the actual wire (see attached for the diagram I used
[image]
)

And I am not sure if I have done this correctly because the print info thing goes on for ever unless I stop it so here is a few lines of what it was producing
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-73
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-72
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-72
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-71
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-70
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-70
h:   0 v:   0 PLL:00 A:7b7b7b S:00.0e.30      I:00 D:0060 m:0 ht:   0 vt:   0 hpw:   0 u:  0 s: 5 W:-70

I've some more questions if you don't mind.  Where can I find more info of what all the options do?  Particularly the Development tab.  For instance what does changing the ram speed do?  I ask as I am still getting jail bar lines and that was oe of my biggest issues with the original board firmware.

And can I just add I am really grateful for you developing this firmware as it really is excellent.

Oh and am I able to install other preset sizez?  My TV is only 720p so I'm not sure if it would look better with 720 x 480 option.  Currently my preset 1 is using the 1280 x 720 option.

ramapcsx2, 2020-01-30:
The info mode lines show that you have an outdated version installed. Always use the latest version ;p  
You should have the source running when taking logs.  

I'm sorry but I don't have the resources to document everything.  
The options in the Development tab are meant for myself, just to quickly test things.  
I leave them available in case they offer what some users may require.  
For the jailbars, it depends on where they come from. I think the Amiga itself is generating them, so there's not much that the GBS can do.  
That really comes down to experimentation though.  

"Other preset size" ?  
You have 5 slots available, and each can hold any resolution.  
Simply switch to a compatible preset, make your adjustments and save to any slot.

rabidgerry, 2020-01-30:
Ok so how can I update the controller with the new version of the software?  I followed the tutorial on here and picked whatever version it mentioned.  Can I dow load the software here also?

ramapcsx2, 2020-01-30:
Always here: https://github.com/ramapcsx2/gbs-control
Use the green "Clone or download" button and get the .zip :)

rabidgerry, 2020-01-30:
How do I do it?  Do I have to basically have to go through the software setup process again?

ramapcsx2, 2020-01-31:
Uh yeah?
It just needs the USB cable plugged in, Arduino IDE loaded with the new sketch, then hit upload and wait a minute.
This is the usual update process.

rabidgerry, 2020-01-31:
Ok basically I just did the whole thing again.  I don't think my preset got saved so I had to create a new one.

The jailbars with this latest firmware seem a little less obvious sothsts good. I don't think they are caused by the Amiga though since a simple RGB to scart cable doesn't create these artefacts. But the unstable picture issue came back.  Had to turn the frame lock option off again to stabilise the image but then I got the tear line again with some games which is disappointing.  I'm not sure why that started up again given it was fine with frame lock on for hours the other night.  I will keep an eye out for the updates in future.  Keep up the good work 😁

ramapcsx2, 2020-01-31:
Okay, so I suppose your unsteady picture problem is just frame time lock not being liked by your display.   
This only has a 50/50 chance of working anyway, all depending on the particular monitor.   
I recently added an external clock generator to the design that fixes this issue, by the way:   
https://shmups.system11.org/viewtopic.php?p=1396532#p1396532

Since the jailbars got better with the update, they are indeed from the GBS.
It helps not to use the 1920x1080 preset, and since you have the GBS8220, you should take a look here:   
https://github.com/ramapcsx2/gbs-control/wiki/GBS-8200-Variants#original-gbs-8220
The ferrite bead might be needed, but I'm not sure.   
I only have one of these boards and it already has the mod. 
There are no jailbars at all with my board :)

rabidgerry, 2020-02-02:
That seems quite interesting, I may buy one of those clock boards.  I'm actually just using my TV at home for using my Amiga.  It's only 720p and 32 inch.  The setting I am using is 1280 x 720 as this was the closest fir for my TV.  Then from here I stretched a little height wise and moved it into the centre.  Since I last commented here the display has been very stable.  However those jail bars seem to be more obvious depending on which colour is on screen.  It's barely detectable with black screen but grey and purple and other it's very noticeable.

I have checked the link you posted before regarding he jail bar mod.  I have actually removed c11 but I put nothing in its place.  I would perhaps think about doing those other mods but to be honest I can't even make out what is going on in the photo regarding R58 and R59.  It just looks like a mass of solder.  If there was step by step pics I'd may be attempt that.

ramapcsx2, 2020-02-02:
Try a different VGA cable, try a VGA > HDMI adapter dongle.

ramapcsx2, 2020-02-07:
For the tear line, see this issue for a hardware fix: https://github.com/ramapcsx2/gbs-control/issues/75#issuecomment-583521088

Jailbar effects are tough to diagnose in text, as there are so many probable causes.   
As I said, another cable could already fix it.

rabidgerry, 2020-02-07:
So you think something like this would solve the unstable screen whilst using frame lock?
[image]
My stability seems to be good enough actually using the framelock.  If it goes weird on my I turn it off and try again, or else turn my tv off and try again and then it stabilises.

The jailbars though are currently pretty bad so they are all that stand in my way between getting a perfect picture using gbs control and gbs 8220.  I was going to make an Amiga rgb to VGA cable to see if that also helps.  I know the pins and what to do, but I am having issues keeping the tiny wires from breaking off the pins.  There is nothing to support them with the rgb hood chassis I am using and no one seems to have documents making one of these cables ever it seems!

ramapcsx2, 2020-02-20:
I prefer these:
https://www.ebay.de/itm/VGA-to-HDMI-Konverter-Adapter-fur-1080P-HD-HDTV-Video-Audio-Box-DVD-PC-Laptop/193308585503

You want as little cabling between the VGA out and the HDMI in, so a dongle like this is perfect.

rabidgerry, 2020-05-02:
Just an update.  I made myself a VGA to (Amiga) RGB D23 cable and used this to go into the GBS8220 via the VGA input.  No real enhancement or benefit other than when I eventually make a case for the board I will be able to connect in a little easier depending design on enclosure.  I had a little trouble at first with this cable but got it working in the end.  I read in the GBS 8220 manual the VGA input is 30.5khz - 32.5khz, so I am wondering how come I do not have any issue with my amiga 15khz output being pushed into the VGA.  Is this perhaps a bad idea to use this method of connecting to the GBS 8220 board?  Just curious as to what you think about that.  This is the reason I initially thought it did not work but that turned out to be a ground wire in the wrong place.