ramapcsx2/gbs-control #103: framesync.h:8:22: error: 'D6' was not declared in this scope

amoore2600, 2020-02-25:
I can not verify/compile gbs-contol for the ESP8266. I get the following error. 

`Arduino: 1.8.12 (Linux), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 4MB (FS:1MB OTA:~1019KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/amoore/Documents/gbs-control/gbs-control.ino:20:0:
sketch/framesync.h: In function 'void risingEdgeISR_measure()':
framesync.h:8:22: error: 'D6' was not declared in this scope
 #define DEBUG_IN_PIN D6
                      ^
sketch/framesync.h:32:19: note: in expansion of macro 'DEBUG_IN_PIN'
   detachInterrupt(DEBUG_IN_PIN);
                   ^
sketch/framesync.h: In function 'void risingEdgeISR_prepare()':
framesync.h:8:22: error: 'D6' was not declared in this scope
 #define DEBUG_IN_PIN D6
                      ^
sketch/framesync.h:40:19: note: in expansion of macro 'DEBUG_IN_PIN'
   detachInterrupt(DEBUG_IN_PIN);
                   ^
sketch/framesync.h: In static member function 'static bool FrameSyncManager<GBS, Attrs>::vsyncOutputSample(uint32_t*, uint32_t*)':
framesync.h:8:22: error: 'D6' was not declared in this scope
 #define DEBUG_IN_PIN D6
                      ^
sketch/framesync.h:68:21: note: in expansion of macro 'DEBUG_IN_PIN'
     attachInterrupt(DEBUG_IN_PIN, risingEdgeISR_prepare, RISING);
                     ^
sketch/framesync.h: In static member function 'static bool FrameSyncManager<GBS, Attrs>::vsyncInputSample(uint32_t*, uint32_t*)':
framesync.h:8:22: error: 'D6' was not declared in this scope
 #define DEBUG_IN_PIN D6
                      ^
sketch/framesync.h:262:21: note: in expansion of macro 'DEBUG_IN_PIN'
     attachInterrupt(DEBUG_IN_PIN, risingEdgeISR_prepare, RISING);
                     ^
/home/amoore/Documents/gbs-control/gbs-control.ino: At global scope:
gbs-control:103:22: error: 'D6' was not declared in this scope
 #define DEBUG_IN_PIN D6 // marked "D12/MISO/D6" (Wemos D1) or D6 (Lolin NodeMCU)
                      ^
/home/amoore/Documents/gbs-control/gbs-control.ino:143:37: note: in expansion of macro 'DEBUG_IN_PIN'
   static const uint8_t debugInPin = DEBUG_IN_PIN;
                                     ^
/home/amoore/Documents/gbs-control/gbs-control.ino: In function 'void setup()':
gbs-control:103:22: error: 'D6' was not declared in this scope
 #define DEBUG_IN_PIN D6 // marked "D12/MISO/D6" (Wemos D1) or D6 (Lolin NodeMCU)
                      ^
/home/amoore/Documents/gbs-control/gbs-control.ino:7008:11: note: in expansion of macro 'DEBUG_IN_PIN'
   pinMode(DEBUG_IN_PIN, INPUT);
           ^
exit status 1
'D6' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Nuck-TH, 2020-02-25:
> Board: "Generic ESP8266 Module
Generic module of cource haven't defines of Wemos pad names.
Choose "LOLIN(WEMOS) D1 R2 & Mini". If you use bare esp8266, you can look at wemos schematics and deduce what pins are used.

amoore2600, 2020-03-13:
@Nuck-TH This fixed my issue. Thank you.

closed 2020-03-13 by amoore2600