ramapcsx2/gbs-control #104: GBS Control with Amiga Screen Tearing

BabCossettes, 2020-03-08:
First off, let me compliment you on a great project! I bought my GBS board a couple of years ago to use with my amiga systems and modern monitors. I was pleased to get it working but before long, the limitations of the original board turned me off from using it. When I discovered your project, I was very excited to give it a try! The results have been fantastic, vastly superior to the stock board and comparable to the high end expensive upscalers available on the market. That being said, I am running into a bit of an issue that I am hoping you can help me with.

The GBS Board I am using is the HC-V3.03 with the yellow buttons. I have removed c11. The cable from the amiga is RGB/CSYNC and for output I am using a VGA to Component cable. 
[image]

For the processor, I am using one of these. A NodeMCU ESP8266 Lua WiFi Internet Development Board - Black from GearBest.
https://www.gearbest.com/transmitters-receivers-module/pp_366523.html?wid=1433363

I installed a version of the software back in May of 2019 (Date stamps on the files are 2019-05-06 2:06PM) which was the best looking one yet! This was back when there was a master and development branch and I believe it was one of the first builds with the scanlines option which I really like. The picture is rock solid in both interlaced and non-interlaced resolutions and extremely stable. Following is an example image and a few print infos from that version of the software.
[image]
h: 427 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2559 vt:271 hpw: 199 s:ff W:31 L:4
h: 427 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05ac m:1 ht:2557 vt:271 hpw: 108 s:ff W:31 L:4
h: 427 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2557 vt:271 hpw: 198 s:ff W:31 L:6
h: 428 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2557 vt:271 hpw: 199 s:ff W:31 L:5
h: 428 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2559 vt:271 hpw: 199 s:ff W:31 L:4
h: 428 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2557 vt:271 hpw: 198 s:ff W:31 L:5
h: 428 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2557 vt:271 hpw: 198 s:ff W:31 L:5
h: 428 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2557 vt:271 hpw: 198 s:ff W:31 L:4
h: 427 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2557 vt:271 hpw: 198 s:ff W:31 L:5
h: 428 v: 525 PLL.00 A:7b7b7b S:8f.10.02 I:00 D:05a4 m:1 ht:2559 vt:271 hpw: 199 s:ff W:31 L:3

The weird thing is, every build since then (I'm one of those guys that always likes to try the latest and greatest), gives me this type of result, a jaggie "tearing" kind of effect that sparkles on the display. Here's a picture and some print infos from the latest version of the software.
[image]
h: 428 v: 524 PLL:0 A:565656 S:8f.10.00 H-   I:00 D:0448 m:1 ht:2345 vt: 259 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 186 u:  0 s:ff S:13 W:31
h: 427 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 186 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0500 m:1 ht:2345 vt: 259 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 259 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 259 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2344 vt: 259 hpw: 186 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 259 hpw: 186 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 187 u:  0 s:ff S:13 W:31
h: 427 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 259 hpw: 871 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 259 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0500 m:1 ht:2345 vt: 259 hpw: 186 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 187 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 186 u:  0 s:ff S:13 W:31
h: 427 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 260 hpw: 186 u:  0 s:ff S:13 W:31
h: 428 v: 524 PLL:0 A:545454 S:8f.10.00 H-   I:00 D:0580 m:1 ht:2345 vt: 259 hpw: 187 u:  0 s:ff S:13 W:31

So, it's not a huge deal because the version I am using from May 2019 works a treat and I'm really pleased with the display. It just would be nice to use new features when they come out and/or maybe somebody else is running into the same issue and it's stopping them from using a truly great solution to upscaling amiga computers.

Thanks again for all your efforts! Hope to hear from you soon.

ramapcsx2, 2020-03-08:
Well, this is actually a good sign. 
The ADC sample rate is close to ideal for the machine, so you get all those perfectly sampled horizontal pixels on the right side of the picture.
Since the phase can not be ideal yet, you also get the bunch of badly sampled pixels on the right though :p 

What you can do is, go to the development tab and hit the "PLL divider++" button a few times.
It increases the sample rate and thus moves it away from the ideal point (which is very hard to maintain, as you see).
At one point, the picture should be good enough, and you can save this as a custom preset.

There are more controls on the development tab that can fix this particular problem, so try them out!

Another thing:
You can only see this issue because you're displaying this checkerboard pattern.
Very little content actually shows patterns like these, so you might be fine if you simply change your "wallpaper" :)

BabCossettes, 2020-03-09:
Hey, thanks for the quick response!

Yeah, that desktop wallpaper is pretty ugly! I actually don't use it normally, just put it up to better illustrate what the issue looks like :-)

The display only does this when I'm using the amiga's interlaced display modes. The non-interlaced display modes look perfect, with no "sparkle".

I tried what you've suggested but unfortunately, no joy yet. I've also tried messing with a number of other settings in development as well but so far the problem persists. I tried switching between motion adaptive de-interlacing and bob. With bob de-interlacing the problem goes away but the display flickers like it used to when the amiga was hooked up to my old 1080 monitor when I ran interlaced displays (just like the good old days!), which was expected based on the description of what the controls will do in the web interface.

I tried turning the gain pots down on the gbs board and if they are all the way to the right, the problem goes away (display is quite dark as expected but still visible enough to see the change).

I'm using a 5V 2A power supply for the GBS board and have the resistor in place on my RGB cable from the amiga. I have an amiga 1000, 3000 and 1200 and they all exhibit the same problem with interlaced displays.

Here's a link to my drive account with a short video showing what this looks like in real time;

https://drive.google.com/file/d/1JR04tI9kL0-s4vyw1OXN29P3v1TyUg1U/view?usp=sharing

Again, not a big deal because that May firmware works perfectly! I'll keep fooling around with the development tab controls and see if I can get any better results. If you have any other ideas, please let me know when you get a chance!

Thanks again for all your hard work!

ramapcsx2, 2020-03-09:
Sure :)

So hmm, that still looks like ADC sampling being near spot on, but changing the deinterlacer mode should have no effect on that.  
Just a quick test: Does changing the SDRAM clock help?

BabCossettes, 2020-03-09:
Nope. Cycling the SDRAM clock changes the appearance slightly but the issue is still there. The output shows Feedback clock, 129Mhz, 144Mhz, 162Mhz, 185Mhz, 108Mhz and back to Feedback clock but still basically the same.

ramapcsx2, 2020-03-09:
Well, one of these days, someone will donate a 500 to the cause.
Once I can work on the problem myself, I'll find a fix :p

BabCossettes, 2020-03-09:
Sounds good! I'll take my cans in and see how much I get! :-)