ramapcsx2/gbs-control #105: Feature request Import/Export presets and separate gain control for RGB.

amoore2600, 2020-03-14:
Is there anyway to import or export presets? 
I have really come up with very accurate and close presets for aspect ratio and screen size for consoles. I'd really like a way to import and export these so they can be shared. I would even settle if there's a way to do this without the web GUI. 

Now that I have really good aspect ratio and screen size I have noticed that I need some tweaks to color from my Gbscontrol. Is there a way adjust RGB, gamma and black levle individually?  Again  I would even settle if there's a way to do this without the web GUI. 

It would be great to have these adjustments and abilities in the GUI if possible.

ramapcsx2, 2020-03-14:
That's all possible, yeah. 
There are some issues though, so let's see.

Preset exporting:
Presets work with a plain, long list of register values, dumped to a file.
They also include flags though, that tell gbscontrol something about the environment where they were taken.
So while they can be dumped and shared, the result are not guaranteed.
This even more so if different gbscontrol revisions are used.
I could add this stuff though, if there is enough interest.
It will be quite a bit of work if the web ui should be involved.

Color controls:
There are quite a number of controls available, from ADC gain and offset to VDS (output processor) fine color controls.
They could be implemented in the web ui, as global overrides, so they apply to all modes where they work (ie: no VDS in bypass mode).
ADC gain is already available.
ADC offsets are calibrated on startup, so they're pretty correct (this is an option in the web ui).
So that leaves the VDS controls: contrast and brightness are the only actually useful things though.
There's nothing for gamma that I know of.

Nuck-TH, 2020-03-14:
Isn't export/import of preset would be as simple as making web server in firmware give out(or save in) preset .txt? I understand, that results may vary(since board electrical parameters vary), but i don't see ways to fix it, so it'd be as-is anyway.

amoore2600, 2020-03-14:
I wanted to show how the colors need to be tuned a bit. 

This test was done on MiSTer using both the HDMI and IO RGBS video sources in real time. 

The GBS-C had the auto ADC Gain applied. Video was scaled and sized as closely as the web gui would allow. 

I have removed the pan pots from the GBS-C  

in the videos below you will see the color shift between sources. 

MiSTer HDMI vs MiSTer IO RGBS GBS-C: NES CORE 240 Test Suite Color bars
https://youtu.be/3OzNRqIlfb0

MiSTer HDMI vs MiSTer IO RGBS GBS-C: SNES CORE SF2WW
https://youtu.be/JV1FsqZbjyw

In other tests with the OSSC using the same set up the color shift are much less pronounced and match near near perfect color wise.

ramapcsx2, 2020-03-14:
Ah yeah, your video shows it perfectly: Your GBS (or something in the chain, but probably the GBS) has at least a red hue to it.
This is exactly what I implemented the auto offset feature for, but apparently it failed here, or it didn't correct enough.
I found this oddity with one of my boards: It persistently requires a rather large ~8% ADC offset change, on the red channel as well, oddly enough.

So that's how you should try to correct it first: Try and change the red channel ADC offset via Serial console:
- g5g09
- hit + or - repeatedly next, it will change the register value :)

It is generally a good idea to correct color issues where they develop, and this is clearly the ADC.
I think I should make a manual offset correction possible for users, seeing this.

ramapcsx2, 2020-03-14:
Regarding the preset compatibility, that is meant for differing software versions.
I assume you want to share presets with others, and that's where it becomes an issue.
Presets could become stale and partly broken.
But yes, it's no problem to access the preset data. Each preset is a single text file on the SPIFFS file system.
If you flashed a SPIFFS server sketch or something similar, you could just download the files.
This is the functionality that I had in mind for the web ui, the one that would take me ages :p

amoore2600, 2020-03-14:
When I try to send "g5g09" I get the following:

Command to send:
`amoore@amoore-OptiPlex-7010:~$ echo "g5g09" > /dev/ttyUSB0`

Monitring the serial terminal:
```
amoore@amoore-OptiPlex-7010:~$ tail -f /dev/ttyUSB0
options set to defaults, restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

startup
(WiFi): STA mode connected; IP: 192.168.0.210
(WiFi): Access 'http://gbscontrol:80' or 'http://gbscontrol.local' (or device IP) in your browser
no ext clockgen
<reset>
Chip ID: 1A 1
 G: 3E G: 3F G: 40
 R: 3E R: 3F R: 40 R: 41 R: 42 R: 43 R: 44 R: 45
 B: 3E B: 3F B: 40 B: 41 B: 42 B: 43
<reset>
Activity detected, input: RGB

2345678
Format change: 1 <stable>
ADC offset: R:45 G:40 B:43
HTotal Adjust: -12, source Hz: 60.099, output Hz: 60.101

preset applied: 1280x960 for NTSC 60Hz 

G5R9 value: 0x7B
unknown command A
```

I am sure I am doing this wrong as my knowledge of arudrino micro computers is limited

ramapcsx2, 2020-03-15:
Oh, it's mostly working. 0x7B is your gain for the red channel. The range is from 0x00 to 0xFF, by the way.
The problem with the unknown command is that your "echo" appends a control character, which the Arduino IDE (or Visual Micro, which I use) doesn't do.
It might still work like that. Simply "echo" a + or - :)

amoore2600, 2020-03-16:
OK, so I have it working but "0x7B" is not gain for red, It's the Horizontal Position:

Shifts the picture left: 
`echo - > /dev/ttyUSB0`

Shifts the picture Right: 
`echo + > /dev/ttyUSB0`

Output from the serial monitor:
```
 Restart
<reset>
Chip ID: 1A 1
<reset>
Activity detected, input: RGB

2345678
Format change: 1 <stable>
ADC offset: R:46 G:40 B:44
HTotal Adjust: -12, source Hz: 60.099, output Hz: 60.101

preset applied: 1280x960 for NTSC 60Hz 

hor. +
hor. +
hor. +
hor. -
hor. -
hor. -
hor. -
hor. -
hor. -

```

Is there any way I find a map of all the registers?

ramapcsx2, 2020-03-17:
Hm, not sure what's wrong. Try directly setting values:
s5s09s60
Maybe using echo makes it not hold on to the segment (5) correctly.

You can find the available documentation in all its glory here (not collected by me ;p)
https://drive.google.com/drive/folders/0B9y2RH4Lb3MZLUYtWGFOMUFVdlE

I'm sorry for the delays. I'm very busy these days, but it'll be worth it! :)

amoore2600, 2020-03-18:
ok I have it working now. 

setting the values directly work with s5s09s60.

Thank you for providing all of the documents. There a little overwhelming. 

I have tried adjusting the RGB gains and offset but I still can't get them perfect. 

I have ordered a new GBS8220 and will test the ADC Gain with it. hopefully with better results. I will let you know what I find.

amoore2600, 2020-03-19:
I have tested with a new GBS8200 and I am still getting the same results. ADC Gain colors are close but not perfect. maybe this is due to the VGA to HDMI encoder I have.

ramapcsx2, 2020-03-19:
Well yea, it's a lot of analog steps to your final image.
There's also all the cabling (especially the connectors) to consider.

But if a GBS always shows a strong shift towards one color, then it's probably internal, and needs to be corrected.
I have only seen one such board, out of maybe 10 or so though.

I'll get the individual ADC gains broken out in the web ui soon!
Remind me if I forget, okay? :p

amoore2600, 2020-04-10:
Sending a soft reminder for the individual ADC gains broken out in the web ui. :-)

ramapcsx2, 2020-04-11:
Hey, I'm still working on that other project. I'll get to this once that's done :)

mastercheff, 2023-02-13:
Well, it's 2023 and the issue is still there. Two boards I have behave similarly - excess of red color...

ramapcsx2, 2023-02-13:
That other project long took over, but maybe someone else can pick this up?
It's about the 3 ADC gain registers, to be added to presets and shown in the web UI.