ramapcsx2/gbs-control #106: A solderless solution for the GBS-control

amoore2600, 2020-03-18:
I have found the 44 pin PLCC to 40 pin DIP adapters. 

https://www.ebay.com/itm/PLCC44-to-DIP40-EZ-Programmer-Adapter-Socket-Universal-Converter-M125/351524231665?hash=item51d87a19f1:g:bGsAAOSwHjNWAT0W

or 

https://www.aliexpress.com/item/32822183074.html?spm=a2g0o.productlist.0.0.36066d1fMDUvxV&algo_pvid=645f582d-e0c4-4dad-b375-dbcc4ea564d9&algo_expid=645f582d-e0c4-4dad-b375-dbcc4ea564d9-1&btsid=0ab6fa8115845586655365171ec46d&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

I am hoping that using jumper wires will move the ESP8226 far enough away to avoid RF interference.  

I happen to have one of these adapter boards and will test soon.

amoore2600, 2020-03-19:
I have tested this and it works well. you can see a video of this here: 
https://youtu.be/q4zKJw4LKFI

I did need to trim down a notch in the adapter to make it fit:
https://photos.app.goo.gl/KMamRYMd7t3dfcZQ9

I also had to trim down the 4 little spaces in the PLCC socket.

I feel like we should be able to get a custom PCB designed so no trimming is needed and has only the pin headers we need.

I had no issues with RF interference when testing.

Shoometsu, 2020-03-19:
I build a PCB myself to use with GBS-control. Some GBS may have
interference with wi-fi signal, but it disappeared when set GBS-control to
connect directly with my router.

wally86, 2020-03-19:
> I have found the 44 pin PLCC to 40 pin DIP adapters. [...]

Anywhere to order in the USA?

ramapcsx2, 2020-03-19:
This looks good at first glance. The distance to the GBS board seems just about enough to move it out of the EMI field. It really reaches quite far sometimes.

One issue with all these sockets is that the connection isn't great, especially over time.
The contacts on the socket and the Myson chip should be cleaned thoroughly, and maybe it's a good idea to glue the socket down (using some not-too-permanent stuff, something like hot glue).

Creating a PCB just for this seems like lots of extra work for little gain.
I think there is a benefit to being able to just buy a readily available PCB from Ali, even if it needs to be modified.
But if you want to, sure :p

closed 2020-03-23 by amoore2600