ramapcsx2/gbs-control #108: Scrambled or no picture with a jamma setup

UnknownMK25, 2020-03-28:
[image]
[image]
 When I hook this up to my jamma set up I either get a scrambled picture as seen in the pic or I don't get any picture at all. It depends on what board is plugged in. The picture will occasionally come in and stabilize, but then just scrambles again. I added the 100 Ω resistor between sync and ground but that didn't make a difference. I also have the pots turned all the way to the left. I tried a 5v 2A and a 12V 2A power supply and that didn't make a difference. Is there something else I can try?

wally86, 2020-03-28:
I'm having a similar issue with my setup. What do the pots do? Mine are all in the middle. I have only used a 5v 2a plug. Can a 12v 2a help at all?

ramapcsx2, 2020-03-28:
No.
Please don't try any more different power supplies. 5V 1 to 2A suffice!

This is a sync signal only problem, and you should investigate what kind of sync your boards put out.
https://github.com/ramapcsx2/gbs-control/wiki/Arcade-Boards

UnknownMK25, 2020-03-29:
It was my suspicion it had something to do with the sync. Can you explain what kind of modification I would need to make to resolve the issue?

ramapcsx2, 2020-03-29:
I cannot debug your setup from here. You need to find out how your board connects to displays.
Do you have a super gun? Did it come with a manual?

UnknownMK25, 2020-03-29:
No manual, it's simply a jamma test rig I built. The rgb is simply coming off the jamma harness and it's just the typical r,g,b,vid sync,gnd, connected here:
[image]
Everything worked fine before the mod and works fine on a CRT, it just seem post mod that the issue is present.

ramapcsx2, 2020-03-29:
I still can't debug this. I don't have access to your setup and can't see how the sync signal looks like.

UnknownMK25, 2020-03-31:
I don't know if this helps any but can you at least tell be if my settings were correct based on my hardware? This is my chip:
[image]
These were the settings I used:
[image]

ramapcsx2, 2020-03-31:
Settings are correct, but I dislike the metal screws. It might be no issue, and surely unrelated, but the metal screw heads could short out stuff.

You need to help me out a little.  
Right now, I only know that you have some arcade board(s) and use the Jamma sync signal directly into the GBS.
It would help if you searched the web for one of your boards with the issue, and find out a bit of info on what kind of sync it may produce.
One important bit is the horizontal frequency, for example. Is it a typical 15kHz signal, or something else?

UnknownMK25, 2020-03-31:
I really can't seem to find anything regarding sync output on the boards I've tested but I *believe* they all output at 15khz and I found this bit of info, not sure if it helps: "Original Arcade boards that run Standard (CGA) run TTL, that means 3 to 5 volts video and sync outputs, terminated at either 390 ohms or 1Kohm". I also spoke to a member of the shmups and klov boards and he indicated that he has run multiple arcade boards on it without issue so there must be something I'm doing wrong.