ramapcsx2/gbs-control #109: Video not scaling and centering

Benny305, 2020-03-29:
Hi there,

Just setup this device and it works great although I have one issue.

The picture by default is not centred and I usually scaled up too much (only part of the video is viewable, like it's zoomed in too much)

I've checked the debug pin is connected to my nodemcu pin D6. A continuity test comes back okay to confirm this with multimeter 

To fix the issue I have to manually scale the picture back and move it to centre it.

Issue happens on all input devices.. (Amiga, N64 - all RGB )

Is this normal or is something not right? 

I assumed the code would auto adjust and scale the image for me.

Only naughty thing I did is run nodemcu from 3.3vcc coming from GBS board. I know this isn't recommended but its power up fine and WiFi reception is great!

Any advice appreciated :) 

Great project! 

Many Thanks 
Ben

ramapcsx2, 2020-03-29:
Scaling in most presets is optimized for full screen display of common sources, such as 90s consoles.  
I can guarantee that default scaling does not remove visible pixels in any of these sources.  

Centering is a function of source, cabling setup, gbscontrol preset and display device.
Any of the factors have an influence, so I recommend you test out options and also check what your display device offers.  
Especially check whether there is an overscan option, and if there is, what it does.

Benny305, 2020-03-29:
Hi Rama,

Thanks for confirming that it should auto scale without removing visible pixels.

In my case, it's scaling in too much and removing a considerable amount of visible pixels.

I shall test the output on a 30khz CRT and also a standard LCD panel and see if the same issue occurs

Cheers

Benny305, 2020-03-29:
Hey Rob,

Well after testing the gbs-control with a 31khz monitor and common LCD panel, I can confirm the picture scales and centers perfectly.

Seems the issue lies with my Epson projector scaling/tracking the VGA signal incorrectly.. so no longer an issue for here :)

Thanks again for the great work you put into this, it's exactly what I was looking for .. a £30 ossc rival!

Just sent you a few "coffee's" to say Danke!

Keep up the good work! 

Cheers,

Ben

closed 2020-03-29 by Benny305

ramapcsx2, 2020-03-29:
Thank you, and good luck with your setup from now on! :)