ramapcsx2/gbs-control #111: C128 RGBi and PC10-II CGA

toms001, 2020-04-01:
Hi,
i connected the GBS8200 with this converter https://github.com/MattisLind/CGA-RGBI-to-Analog-RGB to a Commodore 128 (80 character-mode) to get the right "brown-color".
For a short while a correct picture is displayed, then it gets scrambled and sometimes the gbs-control resets - and again...

I got the same behavior with a Commodore PC10-II connected the same way to CGA. 
The picture is ok, when the Sync Watcher is disabled. Did i miss something?

Thanks and regards, 
toms001

Attached a small log-file: [gbs.log](https://github.com/ramapcsx2/gbs-control/files/4416436/gbs.log)

ramapcsx2, 2020-04-01:
The log tells me that gbscontrol fails to dial in a correct sync pulse window.
The sync pulses from the converter pcb may be particularly short or long, so that they fall out of the wide window margin.

Do you have any timing infos for the converter pcb?

toms001, 2020-04-04:
I think the converter pcb is based on this: https://www.insentricity.com/a.cl/219/cgargbi-to-analog-for-the-commodore-128.

Quote:
Another issue with the CGA/RGBI signal is that it outputs separate horizontal and vertical sync. I hooked them up to the VGA converter but it didn’t work. I found out that the GBS-8200 and knockoffs only work with composite sync on signals that are below VGA frequencies. Mixing the horizontal and vertical signals with a capacitor doesn’t work. Instead I was able to XOR the two signals together with a 7486 which made the VGA converter happy.

ramapcsx2, 2020-04-05:
CSync is definitely easier to process on the GBS, but that's a particular design issue with the scaler chip (one of the units doesn't use the VSync signal, and thus fails to work).

toms001, 2020-04-09:
is it possible, that a optional sync stripper solves the problem?

ramapcsx2, 2020-04-09:
Well, the sync stripper is supposed to separate the CSync signal into H/V Sync, making it harder to work with, but the logs look like it's almost working right now, so you could try it.

Purplegopher, 2020-06-01:
Hello! I recently started looking into C128/RGB [2] and had found and build the same RGBi adapter [1] with a GBS8200. I am having the exact same issue, stable pic for a short while then it switches and changes color balance and switches again after a few seconds.
(wow, that's a long "me too"...)

Is there anything we can do? If it helps, I have the means to give you hardware debug (oscilloscope or logic analyzer), though not much experience there.

Thanks for your amazing software! I hope we can improve it even further :)

[1] https://github.com/MattisLind/CGA-RGBI-to-Analog-RGB
[2] https://mingos-commodorepage.com/tutorials/divrgbconverter2.php?id=170

ramapcsx2, 2020-06-01:
I've recently been given a Neo Geo MVS arcade system.  
It should help tweak the sync window code, or at least help me come up with a different solution to the problem.  
Whatever it is going to be, it will probably affect this issue as well.

Hope you can wait a little longer, until I am finished with my current project.

Purplegopher, 2020-06-01:
Hey there. You owe us nothing, I am grateful that you even do all of this and that this project exists :) I'll wait but maybe we can help. If you have a C128, I could send you one of those circuits (if you don't mind my horrible horrible soldering...) or I could provide data.

I tried the other build. It is stable for a few seconds after boot but then completely freezes the output (the software keeps running) and says "no signal". This is what the log says shows:

```
Restart
<reset>
Chip ID: 1A 1
<reset>
Activity detected, input: RGB

Format change: <stable>
matched preset override > 1280x1024
ADC offset: R:42 G:46 B:44
Phase: 12 SOG: 13
post preset done (preset id: 12) for 50Hz 

.***
no signal
h: 431 v: 623 PLL:01 A:484848 S:a7.00.03 I:00 D:0500 m:2 ht:2269 vt: 311 hpw:2086 u: 96 s:13 TF:0000 W:-55
**************
no signal
h: 431 v: 623 PLL:02 A:484848 S:a7.00.03 I:00 D:0500 m:2 ht:2269 vt: 311 hpw:2086 u:384 s:13 TF:0000 W:-54
******************
no signal
h: 431 v: 623 PLL:03 A:484848 S:a7.00.03 I:00 D:0500 m:2 ht:2269 vt: 311 hpw:2086 u:708 s:13 TF:0000 W:-53
*****
<reset>
```

ramapcsx2, 2020-06-06:
Thanks for the offer!
I don't have any home computers, especially not such rarities :)

It looks like I'll give up on the idea of the entire sync detection being fully automatic.
I can imagine a few (maybe 3) different modes, that can assume different base conditions, instead of trying to auto detect them.
One of these modes should work better, I hope.

But could you do me a favor and post a longer log with "Development > Print Infos" enabled?
It constantly prints the sync conditions, so I can see why it even complains about "no signal" (as the current log looks good otherwise).

Purplegopher, 2020-06-07:
Ok, I tried, but the textarea keeps clearing every few seconds, wasn't really able to capture a full sequence between found/lost/found.

You didn't specify if you wanted the logs from the older or the recent version, so I attached both. I hope they are of use. Of if there's a trick to give you a full output.

[scanlog_20200607_currentversion.txt](https://github.com/ramapcsx2/gbs-control/files/4742667/scanlog_20200607_currentversion.txt)
[scanlog_20200607_oldversion.txt](https://github.com/ramapcsx2/gbs-control/files/4742656/scanlog_20200607_oldversion.txt)

ramapcsx2, 2020-06-08:
Thanks for the logs, puzzling as they are :p 
I think the issue is that the source sync pulses might be active high.  
If you have a way to do this, you could try inverting your sync. I'm not certain it will work, but you could try an inverter logic chip for this.

Purplegopher, 2020-06-12:
Ok, got the scope out and the pulses look like this, 3V peak to peak, active high. Before I try to mess around with an 7406 to invert the logic, is this "in general" what it should look like? Even if it is "inverted"? 15kHz sounds right to me.
[image]

The funny thing is, while the ESP is booting or resetting, the line looks totally different. So the control has some way to mess with the sync pin. Very strange.

ramapcsx2, 2020-06-12:
Yeah, the scope view looks good. It's active high and you should try inverting it.
Try to keep the voltage levels below 3.3V (the assumed limit for the scaler chip).

I also noticed that dip in the sync levels when the scaler enters some particular states.
It's something with the ADC channel for sync, probably draining some signal to ground.

Purplegopher, 2020-06-12:
Ok, I derped around with an 7406 and indeed, stable image!! Thank you, you are AMAZING! :)
I am not quite happy with the sync voltage levels, but that's just me not having experience with all of this.
Thanks!!

ramapcsx2, 2020-06-13:
Nice, good to hear it's working :)
If you want to finalize it, just ensure the unused inputs of 7406 are tied either high or low (so nothing oscillates), and try an inline resistor for the used sync outputs, to bring the signal down a good bit below 3.3Vpp, to be safe.