ramapcsx2/gbs-control #117: White wavy lines?

wally86, 2020-04-26:
Hello all. Got my gbs control finally all up and running properly, but in the background im getting white wavy lines very faint, like interference. I'm using component input. Video cable is fully shielded from retro-access.com. Video input is 240p. I'm using 720*480 resolution in gbs. I'm also using a very thick shielded VGA cable to VGA input on TV. Any help would be appreciated. Thanks

tomudo, 2020-04-30:
Does other output  resolutions have similar problem?

wally86, 2020-04-30:
Yes. Same thing

tomudo, 2020-04-30:
You may try reduce ram clock frequency. This hard mod by rama -> https://shmups.system11.org/viewtopic.php?p=1324748&sid=09ea655e81c7231cf695443691977ceb#p1324748 may be help.

requiem182, 2020-08-14:
I have the same exact problem unfortunately 😔

gingerbeardman, 2020-08-14:
Try a different power source.