ramapcsx2/gbs-control #118: Neogeo MVS sync issue (garbage on screen)

adgenet, 2020-04-27:
I was testing my Neogeo MV-1FS with my GBS board on the most current release and the screen is filled with zigzagging garbage, occasionally showing a somewhat correct looking freeze-frame, then returning to complete garbage zigzagging the screen.
Running the board without gbscontrol shows a slightly unstable picture but it works overall (wanted to test to make sure my hardware isn't broken).
Reverting gbscontrol to an older version I had saved from July 2019 restores a full stable picture. Something has changed between July 2019 (at least) and current that broke Neogeo MVS compatibility.
What information do I need to provide to assist in tracking down the issue?

ramapcsx2, 2020-04-27:
I'm already talking to someone that may be able to send me an MVS.
I need actual hardware to fix this, so please use the older version for now.

adgenet, 2020-05-02:
Thanks for this great project! It's the best thing I've ever come across!
I did some more testing, and found that this is the commit that broke compatibility: https://github.com/ramapcsx2/gbs-control/commit/c364c0ff732518a867fda9dc4d098d9e126956ce
The commit one prior to this (https://github.com/ramapcsx2/gbs-control/commit/abecf525034e820b4ff1be91c41f246cb89ef16c) is the last one that works.

ramapcsx2, 2020-05-02:
Thanks, that'll help fixing it :)

ramapcsx2, 2020-08-26:
I've since received the test machine, but it worked out of the box with the latest revision.

adgenet, 2020-08-30:
Recently installed a clockgen and decided to try the latest revision again.
Still seeing the same behavior.
Maybe different MVS boards (MV-1, F, FS, FZ, FT, A, B, C, etc.) all behave differently?
I initially noticed the issue with my main board, the MV1FS, but I have an MV1B that I picked up since then that I can test with as well. I will report back with results when I have a chance.
@ramapcsx2 What MVS board are you testing with?

blockfeed, 2020-09-03:
@adgenet I have a consolized MV1C that works properly with sync pulled directly from the jamma edge. I have an consolized MV1FZ and will test that as well, hopefully that will offer some clues.

adgenet, 2020-09-04:
I had some spare time today to test with the latest release.
The MV-1F still exhibits the symptoms in the original post.
Strangely, the MV-1B syncs perfectly, much like what you've found with the MV-1C.
I suppose there is a hardware change that happened with the later boards that makes them work. Both are unmodified signals directly from the JAMMA edge.

blockfeed, 2020-09-04:
I had a moment to test my consolized MV1FZ and it also appears to have the issue with sync (pulled from JAMMA edge). I'll see about trying to get some measurements from both my MVS units.

ramapcsx2, 2020-09-06:
@adgenet: Sorry, mine is an MV1B1 and it works just fine.

Spacejockey-GH, 2020-11-02:
I have same problem with MV4FT wih SYNC after installed clock generator and updating to last version. 
CPS2 works great with same configuration.
Older version  (more than 1 year) without clock generator was working perfectly (GREAT JOB Rama). 

 https://wiki.neogeodev.org/index.php?title=Supergun 

"The Neo Geo, amongst most arcade systems, outputs a malformed sync signal without proper serration or equalization pulses. This will not noticeably affect operation most TVs as the horizontal and vertical oscillator aberrations will typically fall within blanked video. Like video, JAMMA sync outputs are high-level and high-impedance. To drive a TV or broadcast monitor's sync input, the JAMMA signal should be buffered, attenuated and impedance-matched. Because the signal is bi-level, a biased logic gate buffer is sufficient. The attenuation/impedance match circuit should have a bandwidth >500 kHz to preserve sync's edges; a typical single-resistor solution forming a voltage divider with the 75 ohm terminator may not be acceptable."

 This is the SYNC present in jamma connector (very ugly by nature)

[image]

And sometimes:
[image]

ramapcsx2, 2020-11-03:
The problem is going to be with some of the newer auto sync window code. This code enables some sources to sync, while it apparently breaks others :(

ramapcsx2, 2020-11-10:
New MVS is underway. It is one of the affected ones, so fingers crossed it leads to a fix :)

claesbert, 2021-03-19:
Hi, Any news on this issue? just tried the latest version (with the very nice ui!), but it's still not working with my (4slot) neo geo mvs. It seems to skip a whole lot of frames and is very choppy. The old version (july  2019) works fine!

Chandler-Kluser, 2025-02-04:
in 2025 and it is still that way, did someone got more information to fix that?

Chandler-Kluser, 2025-02-05:
> Thanks for this great project! It's the best thing I've ever come across! I did some more testing, and found that this is the commit that broke compatibility: [c364c0f](https://github.com/ramapcsx2/gbs-control/commit/c364c0ff732518a867fda9dc4d098d9e126956ce) The commit one prior to this ([abecf52](https://github.com/ramapcsx2/gbs-control/commit/abecf525034e820b4ff1be91c41f246cb89ef16c)) is the last one that works.

I can surely say in 2025 this is still a fact, I tried last commit and it has a lot of frame drops and some glitches....

Building firmware exactly at [this commit](https://github.com/ramapcsx2/gbs-control/commit/abecf525034e820b4ff1be91c41f246cb89ef16c) and adding 470 ohms + 10uF in series with SYNC from MVS JAMMA to GBS Control Sync pin solved the issue!

The sad part is that in this commit I cannot downscale 480p to 15Hz for my CRT television, but the good part is that the image is stable and very nice!