ramapcsx2/gbs-control #120: Horizontal scaling only for VGA

retrorepair, 2020-05-07:
I've been playing around a bit and I've noticed on passthrough mode on my old CRT VGA monitor (maximum resolution 800x600) on passthrough mode only the horizontal resolution would need to be scaled and natural scanlines are already present.

Would this be a feature that could be added in the future?

ramapcsx2, 2020-05-08:
There are 2 operating modes possible here: Full pass-through, which bypasses most chip units, and full processing, which uses all required features.
So while I don't quite understand what you want to do, it doesn't sound like it will work that way.

retrorepair, 2020-05-09:
I'm suggesting a mode that doesn't scale the vertical timing but does the horizontal. So for example 640x480 would be 320x480.

If that isn't possible then I guess it's a moot point.

ramapcsx2, 2020-05-09:
If you want to scale either the X or the Y, you have to use the entire scaling stack.   
So I suppose what you want is an XXXX by 480 preset.

retrorepair, 2020-05-09:
Actually my comment doesn't make sense as I read it back.

So if I use passthrough mode and plug my amiga into my VGA CRT, the vertical "pixels" look correct and produces natural scanlines which looks great, the horizontal "pixels" are smashed togther though as I'm guessing this monitor can't handle a 320 wide sync. So it would actually be 640x240 or something similar.

This I believe is a but of an undocumented "feature" of VGA crts, the video below might explain it a bit better than I can:

https://youtu.be/Y9E3KB0bmLY