ramapcsx2/gbs-control #121: [Feature request] Auto switching Bob/Motion by progressive scan detection.

tomudo, 2020-05-11:
I read description in preference of gbscontrol web UI.

```
Gbscontrol detects interlaced content and automatically toggles deinterlacing.
- Bob Method: essentially no deinterlacing, no added lag but flickers
- Motion Adaptive: removes flicker, adds 1 frame of lag and shows some artefacts in moving details
If possible, configure the source for progressive output. Otherwise, using Motion Adaptive is recommended.
```
From description above, I understand that "bob" is best for progressive and "motion adaptive" should be used in interlaced, right? 

I use gbs control for PS2 game which each games use different interlace / progressive mode. Could you add auto switching bob/motion by interlaced/progressive detection?

Nuck-TH, 2020-05-11:
It is both methods of deinterlacing. If input is progressive it is processed as is.

tomudo, 2020-05-11:
Ok. I got it.

closed 2020-05-11 by tomudo