ramapcsx2/gbs-control #123: 22khz

retrorepair, 2020-05-16:
Is it possibke to scale 22khz for example EGA or the x68000? For some reason I thought it handled this but it doesn't seem to work?

Idskov, 2020-07-19:
Same question but for 19.2khz hsync, 50hz vsync RGBHV.
I guess we can broaden the question a bit to "just" asking for custom hsync frequencies other than the standard supported ones.

ramapcsx2, 2020-07-20:
Each of the sufficiently different horizontal timings need to be dealt with as a new source format.
I handle 22kHz as "medium res" currently, so it should work.

19.2kHz is very low though, so that might fall out of the scope of the mid res code.

It's a lot of work to add support for new timings like this, by the way ;p

Idskov, 2020-07-20:
I was afraid that would be the answer. 
But seeing the board actually output the first frame before freezing, or at least some of it, got my hopes up for it "just" being a tweak. 

[image]

Seems I need to upgrade to a GBS8219 (XVGA) to make it work then.

gingerbeardman, 2020-07-20:
> It's a lot of work to add support for new timings like this, by the way ;p

is there any help the community can give? like a group "resolution timings effort"

ramapcsx2, 2020-07-21:
We could do something with the online capabilities, but I don't know how I could interface such community work with programming the chip.
To "share" the register settings (effectively a preset), the other side at least has to have the automatic routines disabled.  
But other than that, I've no idea how this would work out in praxis...

If you want to try something like this, it would be great to have a bit of the necessary infrastructure ready.   
A small server that can hold a couple preset files (a few kB or so each) and some system to give some metadata to the files :)