ramapcsx2/gbs-control #124: Third-party guide - Add link to wiki?

ObiKKa, 2020-06-07:
Hey. I found this [unique guide page](https://tomdalby.com/other/gbs8200.html) by a Tom Dalby, filled with the steps of what the creator did and photos of a main GBS-8200 board, components, cablings, few diagrams, and many test images from tested hardware machines. It also shows a white custom-made case that he made to assemble around the GBS board and its finished settings, :stuck_out_tongue_winking_eye:.

I think you could add the link as a recommended page in the [hardware setup page](https://github.com/ramapcsx2/gbs-control/wiki/Build-the-Hardware) for other interested learners and builders to have a look at for reference.

Also be sure to go to his [upper-level page](https://tomdalby.com/retro.html) to see the vast collection of his retro computers and consoles (:wink:) that Tom D. uses and tests with his assembled GBS board.

 ---

By the way, I wish that there would be some sellers out there assembling these GBS-8200/8220 boards with cases for sale to users who either _don't_ want to bother with all the hardware and software setups, nor have the required experience with soldering & development board assemblies.

ObiKKa, 2020-06-08:
I have a question. How would you output this thing to component ports on a TV screen, given that the only output port on the board is a VGA port? I had the thought to connect it to a component set on an old Plasma TV. Though, it does have a VGA output port. But component is supposed to be better.

I notice that there are few VGA to component cables for sale on general websites. But their description usually say that they only work with projectors and not normal TVs and monitors, when converting from VGA on a PC. Does the gbs-control firmware make up for that?

This old issue #53 (Dec 2018) briefly details how to build a VGA to component breakout cable. This issue #77 (Aug 2019) was to try to work out how to transcode what seems to be over component output with an interlaced resolution (i.e. 480i) to a TV screen, and also the option of converting from a 240p signal input.

According to [RetroRGB's guide page on converters](https://www.retrorgb.com/converters.html), it seems there is no professional VGA (RGBHV) input to component (YPbPr) output cables/converters. Those are usually just for the SCART inputs only.

gingerbeardman, 2020-06-30:
@ObiKKa in my experience VGA is better, as TVs usually respect the output and leave the signal unprocessed when compared to Component.

General discussion is best at the shmups forum thread: https://shmups.system11.org/viewtopic.php?f=6&t=52172