ramapcsx2/gbs-control #125: Would it be possible to go from VGA to Analog like Composite, YPbPr, etc?

Roulff, 2020-06-15:
I know this might be a long shot but the retro gaming world needs better solutions to connect PCs to Older TVs lol.
Most solutions are too expensive =(
I was wondering if this board could work in reversal, accepting a VGA signal from the PC as in the CRT_emudriver solution, and transcode it to analog like composite and YPbPr.
Would be great if something like that was possible =)

Nuck-TH, 2020-06-15:
VGA is analog (RGBHV) video too :P
VGA input is here, but is better to mix H and V sync, as there is limitations in separate H sync processing.
YPbPr input/output is already possible - you can output resolution as-is, upscale it,  or downscale to 240p.
Composite - no, as TIVA chip lacks both coder and decoder of it.

Roulff, 2020-06-15:
WoW! If that's possible then I'll be ordering one from China right now!
Has anyone already tested this? I really want to get 240p output from a PC to play on my YUV/YPbPr component input TV.
This is going to be much better than other VGA to YPbPr transcoders if it can downscale 480p to 240p without much hassle and lag, maybe the CRT emudriver wouldn't even be needed.
Would really appreciate more comments about this, thanks all.

ramapcsx2, 2020-06-17:
I haven't tested the Component Video output together with downscaling. It'd be great if you could try it :)

richms, 2020-06-17:
I have not been able to get it to convert 720p component down to 240p - best I can get working is a HDMI to compontent box with a scaler taking it down to 480p, that into the GBS and then the GBS downscaling to 240p. Looks very bad compared to the 240p output from my mister direct to the component in on the TV, probably because its been scaled twice. Project is on the backburner for now so hopefully there becomes more options in GBS control before I pick it up again.

Nuck-TH, 2020-06-18:
Maybe you should post details of your setup, as it is hard to make anything from vague description of chain?

gaabora, 2020-10-03:
@richms could you please tell the model of HDMI to component box with a scaler you used or givea link to it?

richms, 2020-10-03:
I got this one https://www.aliexpress.com/item/32808842094.html - it has a button to cycle thru the output formats and the only one that the GBS will take and sucessfully make 240p out is 480p. The 480p from the box already looks quite soft so perhaps a better 480p input to the GBS would work better. Anything higher than 480p into the GBS just seems to get passed thru or have no usable output no matter what options I try. 

It may be better going to RGB but I have to use component out at I only have component or scart in on the TV and no amount of sync combining I have tried on the VGA output has worked for me with the scart input. Nothing else I have that takes VGA will take the 240p signal so at best I can see that its not 240p coming out of the GBS when I put other things into it. Not played for a while because I have deployed a mister FPGA into the arcade shell which is working well enough for me for now, and getting video playback on it from something modern isnt a priority right now.

gaabora, 2020-10-04:
Thanks for reply! And what about latency on that converter from ali? I have a Thompson 32" wide crt tv with scart and component inputs that receive only 480p\576p and I though that GBS will help me to connect any of vga sources in 720p+ easily with downscaling, but unfortunally it does not. So now I'm looking for some other downscaler or maybe will switch to some plasma tv with not so much power consumption.

gingerbeardman, 2020-10-04:
@gaabora I'm also waiting on this type of downscale for my 480p EDTV

i think it would help that when GBS downscale is mentioned it should specify 240p downscale only at this time.