ramapcsx2/gbs-control #126: Proposal for improved Web GUI

gingerbeardman, 2020-06-19:
> This wasn't asked for, so no worries if it's off the table.

I used Bootstrap Studio to put together an improved user interface:
- works better at all screen sizes (responsive)
- features better design language
  - visual tabs
  - different style buttons 
    - momentary (outlined)
    - grouped/separate toggle (filled)
    - danger (red)
  - different style text for notes and warnings

If this revised interface is liked enough to want to merge, I can amend, tweak, and optimise before creating a PR.

I'd like to improve the buttons a bit more, to introduce a set width.

Cheers!

https://www.gingerbeardman.com/gbs-control/

Nuck-TH, 2020-06-19:
currently it is mobile only as it is not even try to fit into desktop screen. and it lacks log console, which is vital for operation monitoring and troubleshooting.

gingerbeardman, 2020-06-19:
> currently it is mobile only as it is not even try to fit into desktop screen

It's most definitely not mobile only.

Can you explain your reason for thinking that?

> and it lacks log console, which is vital for operation monitoring and troubleshooting

Added.

VirtuaVespa, 2020-06-19:
A larger "Freeze Frame" button would be great as it's a bit fiddly to hit precisely on a touch screen

Nuck-TH, 2020-06-19:
> Can you explain your reason for thinking that @Nuck-TH?
All layouting is vertical and doesn't change besides stretching text to the right.

gingerbeardman, 2020-06-19:
> A larger "Freeze Frame" button would be great as it's a bit fiddly to hit precisely on a touch screen

Done, great idea.

> All layouting is vertical and doesn't change besides stretching text to the right.

Much better now.

VirtuaVespa, 2020-06-19:
@gingerbeardman Looks great, I like that it's last on the list too, less likely to hit a button below

gingerbeardman, 2020-06-19:
It can be anywhere. I just moved it to the top so you don't have to scroll to it. It's on a row of its own.

Nuck-TH, 2020-06-19:
Not a demand now but - isn't layout look better if centered, instead of left?
And maybe dark theme? Maybe even with choice and (cookie based?) remembering of it...

gingerbeardman, 2020-06-19:
> isn't layout look better if centered, instead of left?

There is a compromise of having the layout container centred, but the content left aligned.

My choice with vertical buttons are that it takes less searching to find an item in a vertical list, than in a grid.

I will make them alphabetically ordered to help, we'll see.

> And maybe dark theme? Maybe even with choice and (cookie based?) remembering of it...

Great idea, themes are very easy to do with Bootstrap. I'll add it soon.

gingerbeardman, 2020-06-19:
Changes made

- centred container
- more button groups
- button ordering
- dark theme test
- no theme picker just yet

Same URL, remember to force refresh the page to see changes.

ramapcsx2, 2020-06-20:
This looks great, I really like it!
Quick questions:  
Did you check how the old Javascript and the Arduino sketch interact, and do you think this can be kept or adapted?
Can you live with delays on me implementing this? I'm currently too busy to tackle such projects :p

dark-aries, 2020-06-20:
I'd center align everything instead of left align. Also there is too much scrolling in the preferences page. You already have some panels taking up the whole width, put two side by side. Bootstrap should shove them underneath when on mobile. I'd probably also change the move picture and border masking arrows to be aligned like an actual dpad but that's me. Compared to when you first shared this and now, it's already loads better. 

Another thing as well is that the presets are currently shoved side by side with no padding between them. Since they are all one solid colour it doesn't give the impression it's actually a button.

gingerbeardman, 2020-06-20:
> This looks great, I really like it!

Cool!

> Did you check how the old Javascript and the Arduino sketch interact, and do you think this can be kept or adapted?

Yrs

> Can you live with delays on me implementing this? I'm currently too busy to tackle such projects :p 

Yes, same!

gingerbeardman, 2020-06-20:
> I'd center align everything instead of left align. 

Why? What benefits do you think it gives? Just out of interest. 

I'm a believer that left aligned makes things more readable and findable compared to a ragged left that you get with things centred. 

> Also there is too much scrolling in the preferences page. You already have some panels taking up the whole width, put two side by side. Bootstrap should shove them underneath when on mobile. 

I agree. It's on my to-do list. 

> I'd probably also change the move picture and border masking arrows to be aligned like an actual dpad but that's me. Compared to when you first shared this and now, it's already loads better. 

Agreed. I'm working towards this road goal but have yet to find the best way to arrange those arrows. 

> Another thing as well is that the presets are currently shoved side by side with no padding between them. Since they are all one solid colour it doesn't give the impression it's actually a button. 

It's a grouped button, but I see your point. Will try to add some outlines or dividers. 

Thanks for your constructive feedback!

gingerbeardman, 2020-06-20:
Changes
- Preferences page is now 2 columns
  - not 100% happy with this but it is OK for now
- 4-day "d-pad" buttons are in
  - could also have a centre "reset" position?
- Borders on grouped buttons

Same URL, remember to force-refresh the page.

dark-aries, 2020-06-20:
> 
> 
> > I'd center align everything instead of left align.
> 
> Why? What benefits do you think it gives? Just out of interest.
> 

Mainly when there was so much horiztonal free space buttons work better centered compared to left aligned. However now that the most dense page has two columns, I don't see a point of centering anymore after playing around with the test page. 

Also with so much free space and only buttons, center align would mimic a remote in layout. Maybe more useful for mobile at this point. At this point personally I'm pretty happy with the layout at this point.  Minimal scrolling needed on desktop now.

gingerbeardman, 2020-06-21:
Great!

I would like to add (visible or hidden) toggles that are remembered for:

- descriptions (so expert users can hide them)
- log

gingerbeardman, 2020-06-21:
Behind the scenes work for hiding descriptions is done. 

Expert mode toggle will be added soon.

goutsune, 2020-06-24:
I'm mostly happy with redesign, thanks for working on hide-description.
A few suggestions though:
Use some icons instead of custom fonts? I use old android 2 flip-phone and I get long text labels instead of icons, same on my default browser where I don't use remote fonts.

Interactive console? I'm not sure if rework on web-side is enough to do it, but it looks like you can send custom commands already by tinkering with code and inputs. So it would be nice to have input field with history to set custom values, trigger bits and dump presets without need for UART connection.

gingerbeardman, 2020-06-24:
> I'm mostly happy with redesign, thanks for working on hide-description.

:)

> Use some icons instead of custom fonts? I use old android 2 flip-phone and I get long text labels instead of icons, same on my default browser where I don't use remote fonts.

100% yes, this would be part of my optimisations once the design is locked down. Definitely no need for custom remote font for icons when we are only using a handful.

> Interactive console? I'm not sure if rework on web-side is enough to do it, but it looks like you can send custom commands already by tinkering with code and inputs. So it would be nice to have [...]

If this is possible, I can work it in. But I can't do the exploration on how that all works, somebody else would need to do that.

gingerbeardman, 2020-06-26:
Update:
- help button
  - toggles help state across whole UI (no need to toggle on each page)
  - help currently defaults to off/hidden (for testing)
- picture control page no responsive

same URL, make sure to force page refresh 😉

VirtuaVespa, 2020-06-27:
Just wanted to say it's looking great!

gingerbeardman, 2020-06-28:
Couple of todo notes so I don't forget. 

- make momentary buttons have momentary highlight
- disable responsive page zoom on double tap

smartroad, 2020-08-07:
This looks great, I've only just started using this but can't wait for this to be a thing. I was using my mobile phone to adjust for my Amiga and had to move and scale the screen a lot. Double tapping caused the page to zoom in and out, I was going to come on here and ask how to stop that, but i really hope this is released soon!

gingerbeardman, 2020-08-07:
There's no timescale right now, as there's still some work to do. But "eventually" is the best expectation!

ramapcsx2, 2020-08-07:
Right, I didn't forget and we'll push this through asap :)

Arnold101knight, 2020-08-19:
much better. also edit 720x480p to 720/640x480p (im giuseppe from instagram ;))