ramapcsx2/gbs-control #127: 720p mode stretches 4:3 games to 16:9

Em-K-IronFist, 2020-06-23:
1280x720 mode currently stretches games to 16:9 instead of displaying them with black borders on the side like 1080p or 1280x960 modes

goutsune, 2020-06-24:
This is actually intended behavior according to forums. 1920x1080 is intended for TVs, anything else is for monitors, besides it is usually possible to adjust aspect ratio manually.

Em-K-IronFist, 2020-06-27:
In that case, I'm going to edit this to be a feature request, for a new additional preset, because having to take the current 720p preset and squish it back down to 4:3 manually via the Web Interface involves a lot of trial and error, especially with the border mask adjustments

Nuck-TH, 2020-06-27:
Check settings of monitor/tv. Some of them are too smart and ignore resolution in favor to aspect ratio guessed by them.

Em-K-IronFist, 2020-06-27:
> Check settings of monitor/tv. Some of them are too smart and ignore resolution in favor to aspect ratio guessed by them.

I don't have any problems with my monitor detecting the resolution. The problem is the preset stretches the image to fill all 1280 pixels of horizontal width.
My 1080p monitor doesn't have any way to squish this 1280x720 image back down to 4:3; a lot of TVs and monitors still lack this feature, unfortunately. That's why I'm requesting an additional preset.

gingerbeardman, 2020-06-28:
Also, my TV (640x480 VGA LCD) disables resizing controls for its VGA input.

YorkieFl, 2020-07-08:
Got to admit this would be a useful feature for me as well to maintain 4:3 inputs

SephirothFan, 2021-06-01:
I wanted to add to this message chain as well, because I would really like this feature too; as my monitor (even when using an aspect option) puts the 1280x720 image in stretched 16x9 rather than 4x3 like the 1920x1080 option does.

iguanatamer, 2022-03-17:
Yeah, it's really annoying that this is an issue. I'm trying to use this on an old arcade cabinet and 720p stretches the picture to hell and back.