ramapcsx2/gbs-control #128: Green bar and crawling green blobs: 31kHz 480p VGA to 15kHz 240p YPbPr

famiclone6502, 2020-07-13:
When using 31kHz source material like a PC or Dreamcast, there are issues downscaling 31kHz 480p to 240p. Initially the video will not sync, and there are scattered green lines on the left of the image. If you go to the Picture Control tab, you can move the image 1 pixel to the right, and the image will sync properly. However, there is a large green bar on the left hand side of the screen. Occasionally there will be a very distracting bright green "blob" on the left side of the screen that quickly goes from the bottom of the screen to the top.

It has no graphical issues upscaling the same content to a higher resolution.

ramapcsx2, 2020-07-17:
Okay, this is probably caused by the general problems dealing with VGA sources (separate H+V Syncs). 
The GBS has a hardware unit that tells the source format, but it only works with CSync sources.  
Is it possible to have 480p output that uses CSync from your yourses?

famiclone6502, 2020-07-17:
The machine is a Taito Type X, which is basically a PC. However, all of the video options are locked down, which is why I need something like the GBS to connect to a CRT.

ramapcsx2, 2020-07-20:
It would have a better chance with CSync. Maybe you can find and try a sync combiner.
Those would be useful to have for Arcade board owners anyway :)

Example:
https://www.retrorgb.com/vga-to-scart-sync-combiner.html

pgrAm, 2020-10-14:
Just wanna confirm, using an [XNOR type sync combiner](https://www.epanorama.net/faq/vga2rgb/vga2rgbs_ttl.gif) at the input to the GBS gets around this issue

aRUNTU, 2022-07-24:
Hello (too late maybe)
 It is weird that connecting my wii to the GBS-C AIO and hotswapping the wii RGB input to the PC RGB input (720x480@60) allows my JVC TM-H150CG to sync the windows desktop, cheers and thank you.

mysticprysm, 2022-07-28:
A little product I made that will help anyone with this issue. https://mysticprysm.com/product/sync-combiner/