ramapcsx2/gbs-control #131: No Red Connection when taking off the Potentiometers

sethenstien, 2020-07-20:
So I took off the Potentiometers just as shown and now I can't get a Red Signal. Any idea what happened. 
[image]
[image]
[image]

rma945, 2020-12-16:
I have the same issue, after removing the potentiometers I have lost the red color. But now, I have also lost a green color too..

ramapcsx2, 2020-12-16:
Most likely you've shorted a color to ground by accident. Use a multimeter beep test to find out.

rma945, 2020-12-16:
Yes, you are right, Unfortunately, the battery in my multimeter has died, so I can't check shorts at the board after soldering bridges. But now I have tested the board, found shorts and re-solder bridges, so everything is good now. btw, Rama, thank you for that awesome firmware!