ramapcsx2/gbs-control #132: PC EGA / CGA support

claunia, 2020-07-21:
Hi,

I've seen there are some issues talking about EGA/CGA support, but there is nothing in the documentation about wether it works, or not at all, and how to connect the pins, specially EGA's Intensity one.

I think it would be nice if the README included the current support status, future plans, and the wiki gives some information on how to connect them.

Regards

Dethernal, 2020-07-23:
I also have problems with EGA/CGA. I try to build R2R DAC, It actually work, but with some artifacts, right part of image are not captured, small left part are discarded and values for resistors are not optimal. Right now it almost work, but something need to be changed in code to get full image, auto offsets for X and Y and auto width and height scale. looks like horizontal sync frequency are not supported or something.

ramapcsx2, 2020-07-26:
Try working with the development buttons in the web ui.

Dethernal, 2020-07-26:
Of course I tried. With debug, I can see gray region on the left where image was discarded, and on the right just some garbage if I increase scale or move picture.
https://github.com/ramapcsx2/gbs-control/issues/101#issuecomment-591386541
Look at the last photo. One symbol lost on the left, and just green garbage on the right (sometimes not green) and no there are no combination of buttons that fix it. But if I increase output resolution and change oversampling I can slightly change right field, 20-30px maybe, that is it, no more visible picture.

Dethernal, 2020-07-30:
> Try working with the development buttons in the web ui.

If buttons in UI cannot help, is there are something that can change situation in code? Maybe a some piece of code that I can start to fiddling with? I also can change oscillators on GBS8200 if it may be necessary.

ramapcsx2, 2020-07-31:
I think it would be too difficult. The entire processing block needs adjustments to fit those timings.