ramapcsx2/gbs-control #133: sync (?) problem with PAL gamecube

0C-34, 2020-07-23:
I have a PAL gamecube and SNES, with the snes it's all cool and good, but with the gamecube it's just a mess, it's all garbled up.
I can see some times the image, but for a split second and not complete.
For the test i used the exact same cable, because PAL snes and gamecube share the same pinout on the multi-out.
Also the gbs spits out "(debug) ign. length:" even after gamecube is turned off and I have to unplug the power cord to make it work again.
Here the log from the snes, the pal gamecube, and the pal gamecube with a ntsc game( modded, works fine with the crts i have):
https://pastebin.com/vpddCgbf

ramapcsx2, 2020-07-27:
Yes, I remember the Gamecube having a hard to deal with sync signal.  
I think the problem was related to Composite Video termination, so check whether you have the 100 Ohm to ground resistor.
(And if you have it, try removing it.)

0C-34, 2020-08-05:
> 
> 
> Yes, I remember the Gamecube having a hard to deal with sync signal.
> I think the problem was related to Composite Video termination, so check whether you have the 100 Ohm to ground resistor.
> (And if you have it, try removing it.)

hi, sorry if I didn't reply, but in the mean time the gbs i bought became really unstable after using it (too much?) with the aftermarket gamecube cable i bought (opened it later and it was a mess gonna rewire it correctly eventually) so i sent it back to amazon and now i'm waiting for the new one (this time it's the older version with the dual vga output).
i'm gonna definetely update when i mod the new gbs!

0C-34, 2020-09-07:
finally i got the gbs8220 from aliexpress, i modded it and it works flawlessly on the SNES.
Tried the same cable of the SNES but had the same horrible mess i had before.
Then I tried the cable i tried the remade cable (source:[http://members.optusnet.com.au/eviltim/gamescart/gamescart.htm#gamecube](url)) but the 75 Ohm resistor to ground replaced for 100 Ohm because it was the closest i had, result: an even worse mess, nothing is recognizable anymore.
Last thing I tried was removing the resistor but got a blank screen :/
(the gbs still works tho, tested it shortly after with SNES)

0C-34, 2021-05-18:
I close this issue due to not having a setup to properly troubleshoot the broblem. Mainly the problem is the cable, I should get a new scart one, but proper ones are currently only made in UK/US so it's too expensive for me to import them at the moment.

closed 2021-05-18 by 0C-34

ramapcsx2, 2021-05-18:
I'm sorry to hear..
If costs are the issue, can you not build a cable yourself? Can you modify an existing cable?