ramapcsx2/gbs-control #137: return 1080i downscale

ZarameS, 2020-08-05:
With the original firmware i able to downscale 1080i content to 720p 
it's a shame that feature gone with the gbsc 
it's posible to return that feature ??

Arnold101knight, 2020-08-19:
I HAVE THE SAME REQUEST.

but i need 480p downscale too (like stock gbs does) from 720p 1080i/p.

I used it to play on arcade monitor naomi wich works only at 640x480p, with xbox360 and gbs i can downscale all resolutions back to 480p, now instead is not possible, when selecting a HD resolution from the console the gbsc enter in passthrough automatically and can't change resolution. Instead if i choose 480p from the xbox360 alle the resolutions in gbs are available. Simply the downscale isn't working anymore like on stock gbs. 

I'm waiting a reply from the dev please

Arnold101knight, 2020-08-19:
and also add a 480i interlaced downscale from HD resolutions too, thanks

ramapcsx2, 2020-08-20:
Please see here: https://github.com/ramapcsx2/gbs-control/issues/139

closed 2020-08-20 by ramapcsx2