ramapcsx2/gbs-control #138: 1600x1200 mode

3zet, 2020-08-11:
Any chance to add 1600x1200 mode? 
I have nice Dell monitor but I must to use 1280x1024 because there is no option for its native resolution.

Nuck-TH, 2020-08-11:
Even 1080p, according to rama, works with some compromises and on limit of TIVA chip abilities. So no.

ramapcsx2, 2020-08-13:
Yes, basically.
It would be possible to have the resolution, but it would compromise on picture quality. 
The result would look worse than your monitor scaling.

gingerbeardman, 2020-08-13:
Is there a reason it can't accept 1600x1200 for downscaling? (My Extron does)

I downscale games from 1600x1200 to get "free" SSAA from my PC to my 480p display.

ramapcsx2, 2020-08-14:
The downscale feature is very basic. I haven't looked into working with other resolutions yet.

requiem182, 2020-08-15:
I am also using Dell monitor with 1600x1200. I’m looking forward to try  this resolution in the future.

Nuck-TH, 2020-08-15:
Don't. As rama said - you will get worse results than letting monitor upscale lower resolution, so implementing this mode is useless.

richms, 2020-08-16:
Would a 800 x 1200 or similar be workable? That way its the right number of lines so the monitor shouldnt complain and scale it, but the horizontals will just have the monitor sample muleiple times on the same pixel output thinking its 1600 per line?