ramapcsx2/gbs-control #139: PLEASE RETURN DOWNSCALE FUNCTION AS STOCK GBS DOES

Arnold101knight, 2020-08-19:
Problem: I used stock gbs for years to play on sega arcade monitor naomi wich works only at 640x480p, with xbox360 and stock gbs i can downscale all resolutions back to 480p, now instead is not possible, when i select a HD resolution from the x360 (component) the gbsc enter in passthrough automatically and can't change resolution. Instead if i choose 480p from the xbox360 alle the output resolutions in gbsc are available. Simply the downscale isn't working anymore like on stock gbs. 

Also would be great to have a option in the web UI to auto vertical scaling 16/9 games back to full height in 4/3 monitors when using 480p downscale instead of playing with black bars and nstead of pushing 100 times everytime on scaling + button in the web UI. Please.

And please add a 640x480i interlaced output downscale resolution too (from hd resolutions). For using HD consoles on a crt scart television would be perfect. (that don't support progressive scan..)

I'm waiting a reply from the dev please, thanks a lot, i love gbsc.

gingerbeardman, 2020-08-19:
Please add one issue for each request, and it's better if their titles are not all caps.

Arnold101knight, 2020-08-19:
done

ramapcsx2, 2020-08-20:
Okay, gbscontrol is meant to offer improved quality for handling typical retro gaming formats.   
High source resolutions are typically supported by the display directly, but I can see that this won't be available on a 640x480 arcade CRT.   

This is currently not supported on gbscontrol, and adding such presets would be a lot of work.  
I might consider supporting some of the higher source resolutions when I work on the general downscale feature more.   

Until then, please just remove the jumper from the GBS board and let the original software do the job.