ramapcsx2/gbs-control #140: Add a 640x480i interlaced output

Arnold101knight, 2020-08-19:
Please add a 640x480i interlaced output downscale resolution too (from hd resolutions). For using HD consoles on a crt scart television would be perfect. (that don't support progressive scan..) when a 240p is too much low to use on a certain game.

VirtuaVespa, 2020-11-30:
480p > 480i would be great

Roulff, 2021-01-05:
I'll reinforce this wish here.

Being able to downscale a 480p source to 480i, together with the already existing 240p downscaling, would make the GBS-C superb and comparable to other rarer scalers like the extron super emotia.

Would really appreciate it. Thanks for considering.

GutsNukem, 2021-01-26:
I'm part of the few that would really appreciate the ability to downscale to 480i as well, especially for movies directly from a PC.

TyMiles2012, 2021-07-27:
I'd love this too!  With my setup, I can plug my Wii into my component amp, have one output go into my OSSC/mClassic combo for recording/streaming as usual, and another output go into the GBSControl, set it to YPbPr mode, and output downscaled 480i to my consumer CRT

danieljg, 2021-08-17:
I agree that this would be a great option.  Anamorphic 16:9 content can look killer on a 480i set.

vjsindhu-github, 2021-08-19:
Being able to do both 240p & 480i from a 480p source would make this the best for connecting pc's to CRT's.

that1crzywhtguy, 2021-11-30:
If this would be at all possible, I to would find it very useful!

Nordii08, 2022-03-17:
This would be a great addition!

grbolivar, 2022-04-15:
For some content, when downscaling 480p -> 240p details are lost (eg. text may become unreadable). Downscaling to 480i (aka, interlacing the signal) will keep all visual details intact while still allowing the source to be connected to a CRT.

wizardwhosaysni, 2023-04-10:
Adding my own use case to the list :  
SEGA Model 1/2/3 arcade boards output 24kHz, 384p medium resolution.
I want to display this kind of arcade board on a 15kHz CRT monitor.

Current behaviour of GBS Control I'm observing when using 15kHz output :  
Output image only uses about 3 quarters of the screen, vertically, while the last bottom part keeps displaying top part of a frozen frame.
My guess is that for now 384p resolution is simply downscaled by keeping each second line, getting 192p resolution. But I could be wrong and this could be more complex than that.
In any case, at the moment I don't see any viable solution to display 384p on a 15kHz monitor.

Well, for 384p to 240p downscaling a solution might be possible ... with two GBS Control boards in series :  
First one gets 384p and upscales it to 480p, while second one gets 480p and downscales it to 240p. I'm going to give it a try in a few days.  
But as explained above in previous posts, this is not a satisfying solution when 384p could be upscaled to 480i with no loss of visual details.

You could argue that there is another board dedicated to this precise 480i output need : GBS-8100. I also gave it a try and while it works indeed, the image quality is catastrophic : unstable, blurry, and faded colours.

So let's keep this github issue alive until someone either implements 480i output to GBS Control or implements GBS Control for GBS-8100 ! :- )

alopeztorres, 2023-07-01:
It could be really handy, for multiplatform emulation pcs, that has 240p and 480i platforms, for example, if you want to play Nes games.... just 240p is awesome, but on certain situations like ps2, you will find that 240p is not enough for some games that has a really better experience in 480i...... hope this could be a thing!....

RickyL1213, 2023-12-11:
[image]
Not sure if this is the setting to output interlaced mode.

rpf16rj, 2024-04-10:
Any updates? Another example is when using a Dreamcast on RGB mode (pin RGB grounded, but not VGA pin), so dreamcast out 480i on must of games, but using component output only out 240p.

ramapcsx2, 2024-04-10:
Guys, this is something you need to deep dive into as a developer. It won't get solved out of the blue, someone has to sit down, using the register explanations @RickyL1213 posted, comparing with the exisiting code, then bit by bit adjust the existing code to try and implement what the guide says.
You will find that the guide tells you to change maybe 3 register bits, but in actuality, many more registers are related to it and also need changing. Use the exisiting code and comments to figure it out. It will take a somewhat experienced hardware dev maybe 3-4 hours, but someone unexperienced it'll be random :p

ramapcsx2, 2024-04-10:
This task specifically sounds like it will be working mostly on the output, which helps. It should only involve the Display Processor, possibly the Input Formatter, but at least nothing on the ADC or Memory.
This sounds like it will be halving the output line count, while enabling Interlacing, so it might be required to decrease the Display Processor clock speed, or maybe better to find a Clock divider setting and drop that by one step, to make the clock fit.