ramapcsx2/gbs-control #142: Koryuu Transcoder Sync Detection Failure

MrSmiley381, 2020-08-19:
Bug [107](https://github.com/ramapcsx2/gbs-control/issues/107) is marked as closed even though the original issue was never resolved.  In short, the [Koryuu](https://videogameperfection.com/products/koryuu-transcoder/) composite/S-Video -> Component transcoder works as expected with the GBS-8200 in its default operation mode but not with GBSControl.  The original issue conversation history implies that the Koryuu is sending its sync signal slightly out of spec or that GBSControl is not programmed to handle whatever edge case scenario the Koryuu is producing.  Workarounds include using a different composite/S-Video upscaler and feeding it to a GBSControl enabled unit or forcibly disabling the sync watcher until a semi-stable image is produced, though neither sounds ideal.

If necessary I can edit code locally/add debugging code/enable serial debugging at your direction first, then proceed to a screen share if doing so looks to be a worthwhile endeavor.

MrSmiley381, 2020-10-07:
I tried this again with a separate unit, version 5.0, with clock generator installed.  I encountered the same problem, but did find a workaround.  If I disable the sync watcher, the forcibly switch between presets a couple of times, I will have a clean image as expected.  Not marking this as closed but leaving this note here for reference.

wolf-re, 2021-01-14:
I can confirm that I also had problems with sync with my Koryuu (straight from the factory, so most likely fw v1.0). However reading through megari's firmware [CHANGELOG](https://github.com/megari/koryuu-fw/blob/master/releases/CHANGELOG) I found that it lists several fixes related to some form sync problems. I flashed [1.1-RC1](https://github.com/megari/koryuu-fw/blob/master/releases/koryuu-fw_1.1-RC1.zip) onto my Koryuu and this improved the situation.

I do still sometimes get tearing for a short time, which resolves itself after a minute or so. As this tearing also appears with non-Koryuu sources I'm not sure it is related to this issue.

Hope that helps :smile:

wolf-re, 2021-01-23:
So I had a bit more time to test things and my previous post was also missing a bits of info. 

First off the missing info to my previous post: I tried that with an un-modded PAL Nintendo 64 as an s-video source for the Koryuu, and the game was the EU version for Super Mario 64 (NUS-NSMP-EUR). Also I use a GBS 8200 v4.0 with clock generator installed, pots removed, but otherwise un-modified. My GBS works really nicely with my Dreamcast via VGA, no problems so far.

And now some more things I tried: using an [Everdrive64-x7](https://krikzz.com/store/home/55-everdrive-64-x7.html) I played a few NTSC games on my PAL console. Here the console output switches to 60 Hz, but the color encoding coming out of the console is still PAL, so you get what's known as ["PAL60"](https://en.wikipedia.org/wiki/PAL#Multisystem_PAL_support_and_%22PAL_60%22). I'm not 100% sure about the timings the N64 outputs in such a case. However the Koryuu supports that combination and trans-codes it to components. That means the timing should still be what ever the N64 output I think, but the color encoding system doesn't matter any more (it has been trans-coded to components obviously). Feeding this into the OSSC works just fine and gives nice results. However feeding this into my GBS I basically get no stable sync. It constantly tries to find something, and about ever 1~2 seconds the blue LED on the ESP board flashes and you get one correctly captured image, then it is all garbled up (like seen in #107). The output on the status console looks very similar to what [was posted there as well](https://github.com/ramapcsx2/gbs-control/issues/107#issuecomment-607580211).

@ramapcsx2 you said in #107 you would need access to a Koryuu or that setup. I do have a digital scope (4ch, 12mpt, 50Mhz / 1Gs) and can produce captures (of all sides of the chain), would this help you with figuring things out? I also don't mind editing and compiling code and can test things for you. 

It would be super nice if we can improve on GBS control, I sooo much like this piece of software and would like to help :grin:

ramapcsx2, 2021-01-23:
Hey,

I'm time constrained, so I can't work much on gbscontrol right now.
Regarding the syncs and the scope, I meant that I would need the actual hardware here while doing the fixes as in, it's no help if someone else has the tools :p 

It's fine though, as I also have a problematic NeoGeo MVS revision here that I can use to investigate.
All I need is some more time :)

wolf-re, 2021-01-23:
Ah I see your [coffee page](https://www.buymeacoffee.com/ramapcsx2) says you're on xStation right now :+1: 

Well the offer still stands, just let me know :grin:

ramapcsx2, 2021-01-24:
Thanks for the coffee, I really need it these days, lol :)

Well, in the end, the transcoder issue is going to be something like a marginal sync okay window.
These problems can all be fixed, but this stuff really takes a long time to do..

Anatrok, 2024-07-07:
> I tried this again with a separate unit, version 5.0, with clock generator installed.  I encountered the same problem, but did find a workaround.  If I disable the sync watcher, the forcibly switch [...]

This definitely works (had to find sync watcher by enabling developer mode under settings), however you still can't get passthrough to display.  Currently using koryuu and gbs-c to get my N64 s-video to my Morph 4k.  Ideally it would be 240p passthrough and I can use the 4k to upscale.  For now the GBS-C is upscaling to 480p which honestly looks pretty good anyway.

FinnEatsCandy, 2024-08-01:
i'm using a gbs-c and koryuu and have issues with ps1/n64 games running pal60. Before of buying a new scaler i want to know if modding rgb to my pal n64 using n64rgb would fix these issues when directly connecting to scart.

TetrisTyp, 2025-08-31:
Has anyone found a solution for this? GBS-C is great and i use it for everything. Except my N64 cause of the Pal60 Sync Problems.