ramapcsx2/gbs-control #143: Strange diagonal lines on PS2

ferolim, 2020-08-26:
Hi,

When connecting my ps2 on the GBS Control some strange diagonal lines appear as can be seen on the attached image. This happens when using a component cable and also a RGB SCART Sync on Luma cable. 

I was tinkering with the development tab and when increasing the HTotal (1600 to be precise) with the syncwatch off, the lines disappear but the image is still not 100% stable. Another thing that I noticed is that some times the image moves on line down and the diagonal lines disappear but, soon after, moves back to where it was and the diagonal lines come back again.

I'm not using an external clock and neither a sync stripper. I've also tried different VGA to HDMI adapters with same results and two ps2 component cables.

I also have a N64 connected with a RGB SCART Sync on Luma cable and have no problem whatsoever.

Has anybody else had this problem? Is there a known solution or some tips on where on the code I could try to make some tests?

By the way, great job on the GBS Control!

If opening an Issue is not the best way to report this problem that I found, feel free to close it.

Thanks a lot

[image]

ramapcsx2, 2020-08-27:
Quite often the problem is with the power supply to the GBS. Try a different one and see if that does anything.
The recommended mods listed in the Wiki could also help: https://github.com/ramapcsx2/gbs-control/wiki/Hardware-Mod-Library

ferolim, 2020-09-13:
Thanks for the reply! I'll be doing some mods, including a clock generator. Thanks again!

closed 2020-09-13 by ferolim