ramapcsx2/gbs-control #144: Image ratio in 1080p 50hz problem

Romain80200, 2020-08-29:
Hello, on my Pal snes, in 60hz, everything is ok. In 720p with 4/3 ratio or 1080 in 16/9 ration, the picture has the same width.
In 50hz, in 720p and 4/3 ratio, the width is the same as in 60hz.
But in 1080p and 16/9, the width is too big and i can't reduce it with the web interface.

720p:
[image]

1080p:
[image]

Thanks for your works.

Romain80200, 2020-09-03:
Hello. I Can put the display in 4/3 mode but the width is too narrow.
I will try some other game this weekend.
My TV is a Samsung UE32C5100.

ramapcsx2, 2020-09-06:
No no, try putting it into 16:9 / standard wide screen mode. The image should then look like 4:3.  
This is what I meant by "it's baked in" :)

Romain80200, 2020-09-08:
I understand but in 16/9 the ratio is wrong, in 4/3 too.
Too large or too narrow but never the right width.

ramapcsx2, 2020-09-09:
Have you tried looking for more aspect ratio or scaling options on your TV?  
The preset works pretty well here on 16:9 displays.

Romain80200, 2020-09-09:
I need to try with another TV.

jsteel44, 2022-04-25:
PAL SNES at 50hz can squash the image, I'd say this is totally normal. Plug into a CRT and you'll likely see the same thing.