ramapcsx2/gbs-control #146: Vertical jitter with Ambery SDV1

srwalter, 2020-09-06:
I'm using an Ambery SDV1 to convert the Y/C output of my C64 into component video.  This worked reasonably well (had a stable picture at least) with the stock GBS configuration.  With gbs-control the picture seems to jump up and down rapidly.  Here's a snippet from the log, and it looks like vt is jumping between 252 and 261, which is probably not good :-)

h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 427 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 427 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 174 u:  0 s:ff S:14 W:-76
h: 427 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0588 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0580 m:1 ht:2345 vt: 252 hpw: 174 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 252 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0580 m:1 ht:2345 vt: 261 hpw: 173 u:  0 s:ff S:14 W:-76
h: 428 v: 517 PLL:0 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2344 vt: 252 hpw: 174 u:  0 s:ff S:14 W:-76

Any suggestions on what I can try?  FWIW the same gbs-control unit works fine with the 80-col output of the C128, so I don't think it's an issue with my hardware setup.

ramapcsx2, 2020-09-08:
Hm, I assume the converter doesn't output Component Video in the type the GBS expects.  
The sync processor has issues locking on to the vertical sync signal.  
Some code tweaks could fix this, I suppose, but at the cost of lowering compatibility with other sources.

Variables to try and tweak in code:
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L4665

All the code in this bracket:
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L4497
(This is a dynamic window set algorithm that tries to optimize some sync parameters. Try tweaking values or simply get rid of it. Might work!)

srwalter, 2020-09-08:
Thanks for the suggestions!  Re: the second link, I wouldn't expect the video to be interlaced.  The data going into the SDV1 is 240p, at least.  Perhaps that's part of the problem, the SDV1 is "converting" / mangling it to 480i?

srwalter, 2020-09-09:
Not sure what was different, but on one power cycle, rather than jitter, I got a rolling picture.  I.e. like on an analog TV when the V-Hold is off

ramapcsx2, 2020-09-09:
Oh, that's just shorthand for PAL and NTSC like modi. It does mean 240p as well as 480i.  
You'll have a good chance of finding a modification to that part of the code that makes the image stable.

srwalter, 2020-09-13:
FYI I get a slightly different error with a PAL source:

h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-68
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-68
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-68
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0500 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-68
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0500 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-68
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-67
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-66
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-66
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-66
h: 431 v: 624 PLL:3 A:7b7b7b S:a7.10.00 H-   I:00 D:0588 m:2 ht:2269 vt: 310 hpw: 166 u:  0 s:ff S:14 W:-66
h: 431 v: 624 PLL:2 A:7b7b7b S:a7.10.00 H-   I:00 D:0500 m:2 ht:2269 vt: 310 hpw: 966 u:  0 s:ff S:14 W:-67
h: 431 v: 624 PLL:2 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-67
h: 431 v: 624 PLL:2 A:7b7b7b S:a7.10.00 H-   I:00 D:0584 m:2 ht:2269 vt: 310 hpw: 167 u:  0 s:ff S:14 W:-68

I get a rolling screen consistently.  If I bypass the SDV1 and use the composite as Luma, it detects 623 lines instead of 624.  This occurs regardless of my suggested change to PRE_COAST, and other tweaking of PRE_COAST / POST_COAST didn't seem to make a difference

Any suggestions?

ramapcsx2, 2020-09-14:
Hm, it further suggests that the SDV1 processes the sync component, but no other info than that.
You might get it stable by tweaking other variables in that code.

srwalter, 2020-10-01:
So I haven't had much luck with blindly changing variables.  Do you have any general debugging tips?  It would be amazing if I could tell when the decoder *thinks* hsync and vsync are occurring, so I could compare that to reality.  I have a scope and know enough about NTSC/PAL to "see" hsync/vsync, but I have no idea what the decoder is actually doing.  Does it have any debug outputs?

ramapcsx2, 2020-10-03:
I can only give you some general tips:
- you need to hook up the 8266 to your pc and get to a serial terminal (such as the one in the Arduino IDE)
- now you can issue commands such as changing register values
- look through old issues in this project for suggestions on what commands to try, and how the commands work
(example: https://github.com/ramapcsx2/gbs-control/issues/63#issuecomment-479597643)
- before you modify things, you'll want to disable the automatic stuff (syncwatcher) once a picture is somewhat visible
- look for changes in "segment 5" in the chip, which is where the sync processor is set up. For example, coasting setup:
s5s38s01 s5s39s03
    sets segment 5, register 0x38 and 0x39 to pre coast 1 and post coast 3

srwalter, 2020-10-10:
Okay I finally scoped the PAL signal, and it does appear that the SDV1 is messing it up.  So even though it works fine for NTSC, this seems to be a problem with the SDV1, and not gbs-control.  Closing this issue, now :smile:

closed 2020-10-10 by srwalter