ramapcsx2/gbs-control #148: wESP32 — Wired ESP32 support

finos2, 2020-09-19:
Yes.. wifi is cool , but ethernet is more relaibe an prectical . 

i found this pcb , utilizing the ethernet capabiltyes of the esp32 and its intresting 
if this will work with te product . 
https://wesp32.com/software/

[image]

ramapcsx2, 2020-09-19:
Heh, well.. First the project would have to be ported to the ESP32.  
At that point it would have a more refined WiFi radio, so one might check whether that isn't good enough already :)

finos2, 2020-10-11:
well . my bad  i notice carefule and its not a esp32 :-) 
how hard would it be to use the W5100/W5200/W5500-based devices for ethernet ? 
its on the spi bus .

finos2, 2022-05-10:
im coming back to this issue once again . 

could support for W5500 be added ? 
[image]
So we have the option , of ethernet on the esp8266 ?