ramapcsx2/gbs-control #149: Clock generator mod wiki

quantum8, 2020-09-22:
I'm following the notes on the wiki: https://github.com/ramapcsx2/gbs-control/wiki/Si5351-Clock-Generator-install-notes

When getting to the step to connect SDA/SCL, the instructions give conflicting pin numbers to the photo shown.

The instructions say:

    Pin 15 (P3.0/Rxd/HSCL) to Si5351 SCL
    Pin 16 (P3.1/Txd/HSDA) to Si5351 SDA

But looking at the picture it shows:

Pin 26  (P1.1) - SDA
Pin 25  (P1.0) - SCL

Which is the correct one here?

ramapcsx2, 2020-09-25:
I2C is a shared bus. Connect the same SCL that you use on the ESP8266 to the SCL port on the clock generator (and vice versa for SDA).

quantum8, 2020-09-28:
Understood. Makes much more sense connecting directly.