ramapcsx2/gbs-control #150: 1080p preset for 16:9 sources

ZiroGX2, 2020-10-08:
The default preset for 1080p sources seems to squish the image back into 4:3 which is great for my 240p consoles but not so much for my wii. Is there any way to enable 16:9 for 1080p. I would use 1280x960 but my oled forces a terrible upscaling filter through sources below 1080p.
[image]

gingerbeardman, 2020-10-08:
Is your TV 4:3?

Does your TV have a zoom function to squish that image to 16:9?

ZiroGX2, 2020-10-08:
No it is an lg b9 oled. Any zoom options it has makes the image screwed up.

Arnold101knight, 2020-10-11:
I have the same issue with og xbox, when i select 720p setting from gbsc and the input from the console is 480p 4/3 the image is automatically stretched to 16/9, instead if i select 1080p from gbsc the image is much better but only 4/3 same as input, even if i select a game 480p 16/9 supported from xbox the image is still 4/3 in 1080p from gbsc :/

so: 
-add a button from the web menu to select 4/3 or 16/9 stretching for all resolutions supported by gbscontrol
-add support for 16/9 input in 1080p setting in gbscontrol, at moment it outputs 4/3 only even if input is widescreen

Thanks a lot

compucat, 2022-02-21:
Has any progress been made on this issue? 

IMO: this could be solved in the general case by simply adding a toggle to tell GBSC what the input aspect ratio is.

(This assumes that the output is properly displayed with square pixels and not stretched - so 1920x1080 and 1280x720 as 16:9, 1280x960 as 4:3, 1280x1024 as 5:4, etc. 720x480 would be the oddball, since it is natively 2:1 but is ambiguously displayed as 16:9 or 4:3 - curse you, old standards!)

The current behavior is maddeningly inconsistent:

- 1920x1080 is a 16:9 output resolution, but treats all input as 4:3.
  - 4:3 inputs are correctly pillarboxed.
  - 16:9 inputs are incorrectly squished horizontally.
- 1280x720 is a 16:9 output resolution, but treats all input as 16:9.
  - 4:3 inputs are incorrectly stretched horizontally.
  - 16:9 inputs are correctly displayed.
- 1280x960 is a 4:3 output resolution, but treats all input as 4:3:
  - On a 4:3 monitor, 4:3 inputs are correctly displayed.
  - On a 4:3 monitor, 16:9 inputs are incorrectly stretched vertically.
  - On a 16:9 monitor displaying the image 1:1, the same behavior occurs.
  - On a 16:9 monitor set to mangle the image (aka stretch to fill), inverse behavior occurs: 4:3 inputs are incorrectly stretched horizontally, while 16:9 inputs appear correct.
- 720x480 is ambiguously 4:3 or 16:9; behavior matches 1280x960.
- 1280x1024 is a 5:4 output resolution, but treats all input as 4:3; behavior matches 1280x960.

This should really be cleaned up such that as long as the output signal's aspect ratio matches the display's actual aspect ratio, any input can be displayed correctly without relying on the TV/monitor/capture device to mangle the signal!

(Aside: I imagine displaying 1280x1024 as stretched to 4:3 is more common than displaying 1280x1024 natively at 5:4. Still, native 5:4 monitors do exist, and making an exception to assume 1280x1024 is "4:3" defeats the point of making the aspect ratio behavior consistent.)

---
Feature-creep-territory ideas:

- I wonder if it's possible to detect [WSS](https://en.wikipedia.org/wiki/Widescreen_signaling#525_line_systems) on the YPbPr input - if so, it'd be possible to autodetect input aspect ratio :)
- Some people might like an arbitrary input aspect ratio field - for example, displaying 256x240 SNES video in its native 8:7 aspect. This would be useful for other weird and wonderful video modes - computers, rather than consoles, primarily.

R2MGaming, 2022-10-30:
@ramapcsx2 Any update to allow aspect ratio toggling for the GBS-C for both firmware and web GUI? It would be nice to make a preset for 1080p with both aspect ratios (4:3 and 16:9).