ramapcsx2/gbs-control #151: colors are washed out from 480p to 480i input

Arnold101knight, 2020-10-11:
testing og xbox component, when is set to 480p ntsc colors are perfect, when i select pal 480i colors become much worse and washed out. I think is the same with a ps2. Auto gain on brightness is already selected and no change.  gbsc setting is 720p. Why this happen?

ramapcsx2, 2020-10-15:
I think the problem is that sources may send different pixel alignments, or some problem within the scaler chip causes the alignment to switch in the digital processing phase. It's very hard to debug or fix this :/

Arnold101knight, 2020-10-15:
> I think the problem is that sources may send different pixel alignments, or some problem within the scaler chip causes the alignment to switch in the digital processing phase. It's very hard to debug or fix this :/

i tried again the xbox on a crt scart rgb this time, there is no color change from ntsc to pal like i saw on gbscontrol and component cable, instead the loss of sharpening from ntsc to pal is present on the crt too, so it's the console regarding the loss of sharpness in pal mode.

loss of color is only on gbscontrol