ramapcsx2/gbs-control #152: Can the GBS output composite sync?

pgrAm, 2020-10-12:
Hi, I'm using GBS-Control to downscale 480p to 240p for my RGB monitor, however my monitor is composite sync only. Is it possible to get the gbs to output a composite sync?

ramapcsx2, 2020-10-13:
The chip can't output a pure CSync signal, sorry.
There is functionality to embed CSync in the Luma signal, to get Component Video output. 
That is the only way I know of, and of course it requires you to use Component.

If you plan on doing a sync combiner, please share your results! :)

pgrAm, 2020-10-13:
Yeah I did build a sync combiner, 2 actually. I tested with two monitors/TVs (Sony KV1311cr & KV25XBR), The first sync combiner used a "passive" AND gate from [here](https://www.retrorgb.com/building-a-passive-sync-combiner.html), and the second used a 74ls86 from [here](https://www.epanorama.net/faq/vga2rgb/vga2rgbs_ttl.gif). The "passive" AND version worked great on both monitors between the output of the gbs and the input of the monitor, and so did the XNOR. I also used a sync combiner on the input of the gbs because I got poor results down scaling from RGBHV sources. I could not get the GBS to sync using the AND version but it did work the XNOR, I expected this because AFAIK the gbs uses a PLL. 

I'm pretty happy with the results, the sync combiner is a small box in line with the vga cable so its not too bad in terms of extra complexity. Now I only wish I could get the GBS to scale 480p to 480i as well

gingerbeardman, 2020-10-14:
@pgrAm how does downscale RGBHV improve with the sync combiner?

pgrAm, 2020-10-14:
@gingerbeardman, as detailed in [issue#128](https://github.com/ramapcsx2/gbs-control/issues/128), downscaling to 240p from RGBHV sources can have serious artifacts. I don't 100% understand why this is, but it is fixed by using a sync combiner.

Nuck-TH, 2020-10-14:
@gingerbeardman scaler chips has limitations when it works with RGBHV video which re circumvented when syncs are combined and video processed as RGBS.

gingerbeardman, 2020-10-14:
@pgrAm can you post a photo of your setup? I would like to replicate it. Thanks!

pgrAm, 2020-10-14:
@gingerbeardman sure, its pretty simple I think

[image]

The 480p RGBHV (VGA) signal goes in through the blue cable, into the XNOR sync converter (powered by 5v on vga pin9).

[image]

The black cable goes into the second "passive" sync converter (the white box), and into a passive adapter which allows me to plug dsub into the sony RGB multi input.

gingerbeardman, 2020-10-14:
Great, thanks, I'll need the XNOR. Or can I buy one of those online?

Arnold101knight, 2020-10-15:
what is the schematichs or a guide to build the black combiner? passive

gingerbeardman, 2020-10-15:
@Arnold101knight scroll up for links to schematics

finos2, 2020-11-17:
> @gingerbeardman sure, its pretty simple I think
> 
> [image]
> 
> The 480p RGBHV (VGA) signal goes in through the blue cable, into the XNOR sync converter (powered by 5v on vga pin9). Then into the gbs
> 
> [image]
> 
> The black cable goes into the second "passive" sync converter (the clear plastic box), and into a passive adapter which allows me to plug dsub into the sony RGB multi input.

mey i have the schematics of the 2 boards ? (one in the clear box and the other on the scart

gingerbeardman, 2020-11-17:
@finos2 the clear one is linked above: https://github.com/ramapcsx2/gbs-control/issues/152#issuecomment-708020588

twotone8, 2021-02-06:
@gingerbeardman I'm looking to build my own passive RGBHV to RGBS adapter and I'm wondering where you found the hood you used for yours.  It looks quite tidy and I'd like to use it myself.

gingerbeardman, 2021-02-06:
You need to ask @pgrAm

twotone8, 2021-02-06:
Thanks Gingerbearman.  @pgrAm, i'm attempting to build my own passive adapter and was curious where you purchased your HD15 to HD15 connector hood.  It looks quite tidy.

pgrAm, 2021-02-06:
@twotone8, The clear one is 3d printed and the black one I can't recall, it was sitting in my parts bin for a while, but you can get similar ones on mouser, digikey etc.

bob15357, 2021-02-10:
Thank you @pgrAm 
This was the setup I needed.
I used the circuit from passive sync combiner found on the RetroRGB site [here](https://www.retrorgb.com/building-a-passive-sync-combiner.html) and used it before the GBS Control board on the input. On the back output, I have a 15khz passive combiner in got from [here](https://www.arcadexpress.com/en/video-converters/292-1977-umsa-adaptador-vga-15khz-a-scart-15khz.html#/151-power_supply-no) 
This works for downscaling VGA 31khz sources like dreamcast and xbox360 over vga out to my Seleco SMV-290 that takes 15khz Scart 240p input.

santiis2010, 2022-07-30:
> Thank you @pgrAm This was the setup I needed. I used the circuit from passive sync combiner found on the RetroRGB site [here](https://www.retrorgb.com/building-a-passive-sync-combiner.html) and used [...]

I did the same Passive Sync Combiner from Bob web site but is not working for me :|