ramapcsx2/gbs-control #154: No access anymore to GBS User Interface

rabidgerry, 2020-10-20:
For months and months I have had no issue accessing the user interface.  I used it at will with barely any connection issues.  The one problem I had in the beginning was I had to add the device to my router and of course I did this and everything has been fine since.

Now I move house, same internet provider, same phone, same router everything is the same except different house and now I can no longer access 192.168.0.19

Every time I access it just says webpage unavailable.  I also cannot access via the weblinks.  However on my PC I can access those links. but this isn't really any use as the whole point is I have access via my smartphone.

Any ideas what has happened?

gingerbeardman, 2020-10-21:
What WiFi access points are you seeing? Maybe GBS-Control has gone back to default?

How long are you leaving the device in for before accessing? The web UI can take a while to become active.

rabidgerry, 2020-10-21:
From where, my phone?  I used to see the GBS control as a wifi option.  But once I added my GBS to my wifi router (I had to log in and do this from my routers IP and gui interface) my phone no longer sees the GBS as a wifi device and I just go to the 192.168.0.19 to access the interface.

However this is no longer working.

I checked my router and the GBS control is still on it.

Also I have had the GBS on for a while so time should not be an issue.

ramapcsx2, 2020-10-25:
Try http://gbscontrol or http://gbscontrol.local (type it manually in your browser address bar).
In case there is WiFi disturbance, leave your console turned off, so only the GBS is powered.

rabidgerry, 2020-10-26:
Ok well there has been a development.  I logged into my router again and had a look.  I was thinking why not try remove the GBS control and then add it.  It was then I noticed that the IP of the GBSC had changed.  So it was now working from from 192.168.0.23/24.  However when I took it into the room where I intend to keep I could no longer access it.  So I am thinking a wifi extender may solve this secondary issue.  The GBSC need to be in range of the router yes?  

Unrelated issue, I tried to join this forum a few months ago but I believe I was never approved my a moderator, Rama do you know any moderators on this forum?

ramapcsx2, 2020-11-01:
Oh, you mean https://shmups.system11.org ?
I don't know mods there off hand, but normally people can join just fine.
You'll have to wait a few days before your first posts go live though, I think.

For the WiFi, the ESP8266's antenna is normally good enough so that you can move the thing a bit away from the router.

rabidgerry, 2020-11-23:
> Oh, you mean https://shmups.system11.org ?
> I don't know mods there off hand, but normally people can join just fine.
> You'll have to wait a few days before your first posts go live though, I think. [...]

Months and months ago I tried to join that board and no one ever approved my account.

Back to the issue.

I got a wifi extender for my house and it solved the issue of connecting to the GBSC GUI.  I'm not sure if my phone was having issues connecting to my wifi from where I have GBS located or if it was the Arduino board has issues.  Either way I can now connect.

lovrenco, 2021-01-29:
@rabidgerry Please close the issue if it was solved. Thank you.

closed 2021-01-29 by ramapcsx2