ramapcsx2/gbs-control #155: Black screen with Playstation at 50hz pal games, no problem with ntsc games

Arnold101knight, 2020-10-22:
i have the same issue found when selecting 1080i resolution on original xbox (black screen)

same problem appen with playstation, all ntsc games are fine, any pal game 50hz give a black screen.
changing output resolution not fix it, not even selecting force 60hz on pal in gbsc
Please fix :(

ps. when a new firmware will be released? 

Thanks!

Arnold101knight, 2020-11-13:
I tested a pal ps1 too a 7502 modded to csync, i can't even see the menu!!! The developer say that works fine...so why for me not work at all 50hz sources? I have a full hd led tv. Please i need help here thanks

ramapcsx2, 2020-11-13:
Gbscontrol generates 50Hz outputs just fine. It is probably your TV that simply can't do 50Hz.
There is an option to force PAL to 60Hz in the web ui preferences tab.

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
Im trying a PAL PS2 set to ybpr and component cable, black screen...
I choosed force PAL 60hz in gbsc and rebooted but not fixed

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
Output Format: Component
pal forced 60hz: apply vshift
ADC offset: R:41 G:43 B:45
clock gen reset: 108000000
Output Format: Component

preset applied: 1280x720 for NTSC 60Hz 

2345678
Format change: 2 <stable>
pal forced 60hz: apply vshift
ADC offset: R:41 G:43 B:45
clock gen reset: 108000000
Output Format: Component

preset applied: 1920x1080 for NTSC 60Hz (set your TV aspect ratio to 16:9!)

source Hz: 50.00028 new out: 50.00028 clock: 90190968 (-17809032)
pal forced 60hz: apply vshift
ADC offset: R:41 G:43 B:45
clock gen reset: 108000000

preset applied: 1280x720 for NTSC 60Hz 

2345678
Format change: 2 <stable>
pal forced 60hz: apply vshift
ADC offset: R:41 G:43 B:45
clock gen reset: 108000000

preset applied: 1920x1080 for NTSC 60Hz (set your TV aspect ratio to 16:9!)

source Hz: 50.00031 new out: 50.00028 clock: 90191024 (-17808976)
Deinterlacer: Bob
Active FrameTime Lock not necessary with external clock gen installed
FTL on
Active FrameTime Lock not necessary with external clock gen installed
pal force 60hz  Restart
(WiFi): AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
<reset>
Chip ID: 1A 1
<reset>
Activity detected, input: Component

2345678
Format change: 2 <stable>
full height
ADC offset: R:40 G:40 B:40
clock gen reset: 81000000
HTotal Adjust: -7
Active FrameTime Lock not necessary with external clock gen installed

preset applied: 1920x1080 for PAL 50Hz (set your TV aspect ratio to 16:9!)

source Hz: 50.00028 new out: 50.00028 clock: 80957984 (-42016)
pal force 60hz  Restart
clock gen reset: 90191024
ADC offset: R:40 G:40 B:40
clock gen reset: 90191024
pass-through on
pal forced 60hz: apply vshift
ADC offset: R:40 G:40 B:40
clock gen reset: 108000000
Active FrameTime Lock not necessary with external clock gen installed

preset applied: 1920x1080 for NTSC 60Hz (set your TV aspect ratio to 16:9!)

2345678
Format change: 2 <stable>
pal forced 60hz: apply vshift
[... 133 lines cut ...]
pal forced 60hz: apply vshift
ADC offset: R:40 G:40 B:40
clock gen reset: 108000000
Active FrameTime Lock not necessary with external clock gen installed

preset applied: 1920x1080 for NTSC 60Hz (set your TV aspect ratio to 16:9!)

source Hz: 50.00031 new out: 50.00031 clock: 90191160 (-17808840)
pal force 60hz  Restart
(WiFi): AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
<reset>
Chip ID: 1A 1
<reset>
Activity detected, input: Component

2345678
Format change: 2 <stable>
ADC offset: R:40 G:40 B:40
clock gen reset: 81000000
HTotal Adjust: -7
Active FrameTime Lock not necessary with external clock gen installed

preset applied: 1920x1080 for PAL 50Hz (set your TV aspect ratio to 16:9!)

source Hz: 50.00031 new out: 50.00031 clock: 80958136 (-41864)

Arnold101knight, 2020-11-13:
Even if i click force PAL 60hz and then reboot gbscontrol It says again PAL 50hz in the log.

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
My TV works fine with HDMI 50hz. Just tested with a xbox360 see photo

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
[image]
  

my setup

Arnold101knight, 2020-11-13:
[image]

this is what i have after selecting force pal 60hz and reboot

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
[image]

Arnold101knight, 2020-11-13:
[image]

whats wrong with clock?

Arnold101knight, 2020-11-13:
i just found the problem...is not the 50 hz! i tried a ps2 game in pal 60hz on crt i see image and on gbscontrol not, so it's the PAL mode the problem. Cam be related to the clock gen or not?

Arnold101knight, 2020-11-13:
i tested on a pc  monitor too, not sync. video: https://youtu.be/eYeFPTr31oU

please help me fix it

ramapcsx2, 2020-11-13:
Sorry, you need to figure out what's wrong with your setup.

Arnold101knight, 2020-11-14:
> Sorry, you need to figure out what's wrong with your setup.

But if you not help me in where i have to check how i can ever fix It? If you need detailed photos of how i have soldered things i can do It because i definetly want to fix It.. please let me know Robert

Arnold101knight, 2020-11-14:
without the vga to hdmi adapter i have a image on the pc monitor, BUT i have an error from the osd of the monitor, that the kz and hz are wrong, there is a problem with pal mode on gbscontrol and that problem make go crazy the hdmi adapter too. 
And i try to set force pal to 60hz and then reboot gbscontrol but the log in gbs control says 50hz -.-  It's bugged or what?

pal ps2 with component cable:
[image]

Arnold101knight, 2020-11-14:
Can you help please? Why the fuck im the only that have those problems with PAL sources? I want to finally fix It and understand where the error Is because im getting crazy for this

ramapcsx2, 2020-11-15:
I'm sorry but look at your picture of your monitor. Read what it says. There is no error, the signal is fine. Your monitor can display it but it warns you about the 50Hz timings. It wants you to switch to 60Hz.
Your GBS works, PAL works. It's your displays.

Arnold101knight, 2020-11-15:
> I'm sorry but look at your picture of your monitor. Read what it says. There is no error, the signal is fine. Your monitor can display it but it warns you about the 50Hz timings. It wants you to switch to 60Hz.
> Your GBS works, PAL works. It's your displays.

Is not true because with the HDMI adapter there Is no Sync at all. Even the HDMI adapter not works?? Its the same model that uses retrorgb. 
The last try i have to do Is use a VGA to component cable and test It on the TV, test a old CRT monitor in VGA (even those monitors not accept PAL 50hz??...) and try to disconnect external clock board.

Also you told before that was my TV that not accept PAL 50hz and its not true because i showed that with xbox360 works fine.

Also why i set force PAL to 60hz in gbscontrol and then reboot It and again in the log say 50hz?! Even that Is normal??

gingerbeardman, 2020-11-15:
I'll test PAL from PS2 on my TV soon.

Arnold101knight, 2020-11-15:
> I'll test PAL from PS2 directly on my TV soon. My TV does not show Hz sync, but we'll see if it works.

Ok thanks

ramapcsx2, 2020-11-15:
The PAL at 60 mode already exists.
Everything you have trouble with works for me, and many others.
I can't fix what's not broken.

Arnold101knight, 2020-11-16:
> The PAL at 60 mode already exists.
> Everything you have trouble with works for me, and many others.
> I can't fix what's not broken.

1. PAL to 60hz option in gbscontrol not works to me, in the log It say 50hz after rebooting It...never worked for me.

2. I not mean PAL 60 hz but output PAL to NTSC 60hz instead

craigarty, 2021-06-09:
Hi I have same problem with UK Pal Dreamcast and Mega drive. NTSC j PC engine works great, but cannot get the Pal devices to work with VGA to HDMI adapter (looks the same as above photo). Connected the VGA port from GBS into a TV VGA port and worked perfectly. Guessing that those cheap HDMI converters don't like outputting PAL. Does anyone know of a good converter that will definitely output PAL 50?

Thanks

Arnold101knight, 2021-06-09:
yeah, interest me too. i thinked was that shit. 
Good job bob from retro rgb for choosing this non pal compatible shit. Who cares of non usa peoples of course

ramapcsx2, 2021-06-10:
The problem might not be with the HDMI dongle, but the TV not accepting 50Hz over HDMI.
If you have another 50Hz source, does that work?

craigarty, 2021-06-10:
I will try it with my older pc vga and set it to 50hz. But being a PAL TV it should support 50/60 Hz through hdmi. The TV is an early 4K Samsung when they first dropped in price to the £1200 mark. It still has Scart RGB so I know the cables work it is just the picture and timings that look crap, the GBS Control is so much better.

Arnold101knight, 2021-06-10:
why you deleted my comment? no words. because i nominated retro rgb Bob who have used this same converter that not works on pal?

and i already stated months ago that my tv works fine at 50hz, so its obviously the dongle

craigarty, 2021-06-12:
Just synched my Dreamcast selected force 60hz, changed low Res up scale off and then on and it syncs. Try megadrive next.

craigarty, 2021-06-12:
Update, game has to be set to 60hz input to keep sync. Just have enough time to select 60.

craigarty, 2021-06-13:
Megadrive working, even did the switchable region mod. 50/60hz fine but only works with1025 or 1080 resolutions.