ramapcsx2/gbs-control #156: Preset 720x480 on 4:3 crt pc monitor

Solidzera, 2020-11-01:
Hi I am using the amazing GBS Control with a pc CRT monitor 4:3 LG 17" 710E with PS2 and Wii through component cable.
When I use 480p game on PS2 or Wii on 480p mode and use GBS Control on Passthrough mode the pc monitor use full horizontal width without black border on the right and left sides of the image.

But when I use 720x480 mode on GBS Control on 480i/240p games the pc monitor gets black border on the right and left sides and the horizontal width is set at max on the monitor already.

I tried to use scaling on GBS Control but I get geometry problem verified on 240p test suite.

I have noticed that older GBS Control version had 640x480 mode instead of 720x480.

There is a way to solve this? Thanks in advance for this amazing project!

ramapcsx2, 2020-11-01:
Hey, this basically falls under adding another preset.
I don't want to do that anymore, since each preset multiplies maintenance effort.

But instead, please try the Picture Controls functions. Maybe you can find a good setting, then save it as a custom preset.

The behaviour of your monitor sounds a bit weird, by the way. The timings are correct to fill the entire width of the screen.

Solidzera, 2020-11-01:
No problem Rama! 
I will do a new custom preset for it.

Solidzera, 2020-11-01:
When I use passthrough on 480p input signals no problem at all. Only when I use 740x480 or 1280x960 preset i get black borders on the right and left sides.

No problem Rama! I am going to create a new preset for it with Picture Controls on GBS as you recommend. 
Continue with your great work!!

closed 2020-11-02 by Solidzera

gingerbeardman, 2020-11-03:
Is it possible to share the preset? I've been wanting this also.

Solidzera, 2020-11-05:
@gingerbeardman I made a custom preset using Picture Controls
Is this you want?
[image]

Solidzera, 2020-11-05:
Also @ramapcsx2 I bought a board that came with a switch for select rgbhv or component circuit for output.
Is there a problem or it is safe using component output from this circuit for crt pc monitors?
I noted the image is even brighter over component than over rgbhv 
And is even better somehow

gingerbeardman, 2020-11-08:
@Solidzera thanks! also can you share where you bought your board? thanks!

Solidzera, 2020-11-09:
@gingerbeardman from this amazing people in Brazil:
https://gamescare.com.br/produto/gbs-control-gamescare-scaller-1080p/
I recommend their products!