ramapcsx2/gbs-control #159: Video input source

finos2, 2020-11-18:
hello . 
can  multiple inputs be connected and change input via software ? or do i have to manualy disconnect and reconnect sources ?

gingerbeardman, 2020-11-19:
Only one source can be active at once. I use a KVM VGA switch to get around this.

finos2, 2020-11-19:
@gingerbeardman  and what about RGBs switching between and RGBHV in my setup i want to have my VGA source (dos/win98 pc)  
simultaneously connected (but not both runing ) with my RGBs source , what do you think is the best way to handle this ? (add to the mix my ps2 connected via ypbpr

gingerbeardman, 2020-11-19:
I use VGA for everything, here is my setup.

- Wii, PS2, Xbox going from Component/YPbPr to VGA (via adapter)
- Windows PC going from HDMI/DisplayPort to VGA (via adapter)
- Xbox360, DC to VGA (dedicated cables)

All going through a TRENDnet TK-409K KVM Switch, before reaching the GBS. 

After GBS I also run the VGA output through an Extron so that the alignment differences are unified, and my 480p VGA LCD TV is locked to this signal. Final output is over DVI.