ramapcsx2/gbs-control #160: Clock Generator Mod - pin 40

rabidgerry, 2020-11-23:
When I first heard about this mod I didn't really consider it, however I have since changed my mind.

However looking at the Si5351 Clock Generator install notes it looks a little different from how I initially remember the mod being done.

I seem to recall a pin 40 leg had to be lifted off the main board and soldered to the clock gen board.  But looking at the install notes I don't see anywhere that says "lift the pin 40 leg".  I recall people where discussing all kinds of techniques for lifting this leg as well, one included thing fishing line that once the solder was heated allowed the person to gentle pull the fishing line to lift the pin 40 leg away from the pcb.

Did I imagine this or has something changed?

gingerbeardman, 2020-11-23:
You can see the revisions of the "Si5351  Clock Generator install" wiki page:

https://github.com/ramapcsx2/gbs-control/wiki/Si5351-Clock-Generator-install-notes/_history

Wiki was changed June 16th to reflect no need to lift the pin.

rabidgerry, 2020-11-23:
EXCELLENT!

Forgive my ignorance but what advantage does the clock mod offer mere mortals other than quicker stabilized picture?  Just curious, I am still going to carry out the mod.

ramapcsx2, 2020-11-23:
Hey,
the clock mod does exactly one thing: It removes the "tear line" that otherwise wanders across the screen occasionally.

rabidgerry, 2020-11-24:
I see.  No picture improvement otherwise?  

Sometimes with scrolling games (on my Amiga 1200) I get like a light smearing effect and I dunno if it's my monitor or my GBS.  I dunno if this what is described as lag.  I know the CRT I have with GBS does the same thing as my main LCD monitor.  Lots of games are perfect and then others will have a sort of light distorted thing happen whilst moving left or right.  It is more the background this happens with.

ramapcsx2, 2020-11-24:
The tear line is such a "light distorted thing". It would move up or down the screen and will only be visible when the background scrolls horizontally.

rabidgerry, 2020-11-24:
hmmm I thought a tear line was just one line though that would move (in my case when I have them occasionally up the scree) up or across.

Ok perhaps I will try this mod and see if I notice anything better with the picture.  Thanks for helping it is always appreciated.

rabidgerry, 2020-11-25:
If I am connecting pin 40 to CLK0's middle pad, do I have to use enamelled wire?  I was thinking I could use a strand of copper from say a thicker wire.  Would that be possible are are their very specific reason as to why enamelled wire is suggested?

ramapcsx2, 2020-11-25:
The requirement is a thin and short wire, and ideally it should have some protection.
If you take a copper strand, the requirement is met, but it's possible the wire could short something if you're not careful.

rabidgerry, 2020-12-11:
Time to perform this mod!  So one last thing, I have located some enamelled wire, it is 0.15mm, should this be thin enough?

Am I right in thinking enamelled wire is to protect the wire from shorting?  Almost like it is insulated?  It solders as normal though right?  Once I know I will hopefully complete the mod and document as best I can for others to hopefully benefit from.  Thanks in advance.