ramapcsx2/gbs-control #162: Xbox 720p - No image

Roy12321334, 2020-12-04:
I have a GBS-control + clock mod.
Setup is an Xbox (2001) w/ component cables -> GBS-C -> Sony CPD-G520 CRT Monitor

The setup works fine when I use 480i and 480p but 720p gives me a black screen.

...
2
2345678
Format change: 5 <stable>
clock gen reset: 161998784
ADC offset: R:48 G:46 B:46
clock gen reset: 161998784
pass-though on

gingerbeardman, 2020-12-05:
Related to #155?

ramapcsx2, 2020-12-05:
720p sources are set to use pass-through, and according to the logs, the scaler is doing that.
When I tested the XBOX, I got a good image from it in 720p.
Sorry I can't say more. It was working here ><

Roy12321334, 2020-12-05:
> 720p sources are set to use pass-through, and according to the logs, the scaler is doing that.
> When I tested the XBOX, I got a good image from it in 720p.
> Sorry I can't say more. It was working here ><

Is there any way to use scaling + 720p? 

I flashed my Xbox with the VGA SoG BIOS and now all resolutions work fine (480p/720p/1080i).

ramapcsx2, 2020-12-06:
The chip is good at upscaling low res, but making 480p look good is already a struggle.
720p would be way too much, and any scaling preset for it would look bad.