ramapcsx2/gbs-control #164: 480p has wavy lines where 720p/960p does not

amerika13, 2020-12-05:
I recently built my GBS-C and it seems to be working quite nicely. However, I noticed an oddity with it when I put it in the 480p preset mode. Moving lines are seen when I use the 480p preset. However, when I use 720p or 960p those lines disappear.

My setup:
1. GBS-C with clock sync generator board and all of the required and most of the recommended mods for an 8200 board. I also have MobiusStripTech's sync combiner board installed but it is only for VGA input issues with systems like the x68k and Naomi.
2. HD Retrovision component cables (SMS and Saturn tested so far) and I also have used scart to component (shinybow) converter with amazon basics component cables for my DC. Both show the same lines in 480p.
3. VGA to HDMI converter (active and it's powered with audio in)
4. HDMI splitter. One end goes to my LCD TV and the other to my Micomsoft internal capture card.

Here are a couple of YT videos showing the issue:
https://www.youtube.com/watch?v=RNmey20n80o

https://www.youtube.com/watch?v=57vsJYqI1Ss

I can give you install pictures, game image captures, video captures or any other info needed.

Thanks

gingerbeardman, 2020-12-05:
Use a different power supply, higher quality is better. For example Apple iPhone USB adapter.

Or go straight to the wall rather than extension cable.

Try many things, you'll find your own answer.

amerika13, 2020-12-05:
I will try more PSU's.  That doesn't explain why the lines are there in 480p but not there in the other modes unless you want to explain further @gingerbeardman

amerika13, 2020-12-05:
I just tried with an Apple branded USB charger and an Android branded USB charger. Both have the same issues with the wavy lines in 480p mode yet are totally fine with any other preset.

gingerbeardman, 2020-12-07:
Interesting, I use an Apple branded USB adapter plus USB to barrel cable: no wavy lines.

The only other variable could be position of Arduino board relative to GBS. Here is mine: https://www.instagram.com/p/CCNvoRUnQMC/

How is yours situated?

amerika13, 2020-12-07:
My Arduino is attached to my GBS board and near the component in RCA jacks. I understand what you're trying to ask, is the Arduino, and probably specifically the WiFi chip, causing interference with analog in. However, just like with testing power, it doesn't make sense because the interference is only happening with the 480p preset. If it was power or arduino interference, it should be seen at all presets. Unless the Arduino is doing something very different while set to 480p than the other settings. Which, in this case, it has to be.

gingerbeardman, 2020-12-07:
Sure. Think of it another way: you are not the first to report this, just the most recent. The cause and solution have been known for a long time and many other people have resolved the issue successfully. Including me: I had this very issue with my first PSU, didn't have it with the Apple USB plug, and don't have it with my current extension cable with USB sockets.

Reference official support thread: https://shmups.system11.org/viewtopic.php?f=6&t=52172 (you can use forum search with search term like: GPS 8200 diagonal)

amerika13, 2020-12-07:
I appreciate the attempted assist @gingerbeardman, but you do not seem to understand the difference between my situation, which I have shown video of, and your own/others. I did already search and the others that reported interference issues said it was an issue with all resolutions, and not just one in particular. That is what brought me here in the first place as my issue seems unique as it doesn't appear to be basic/common interference and something Rama might want to see.

Thank you for the link to the official support thread. I will bring my issue up there to see if anybody else can chime in. Also, I did try searching there as well and came up with nothing.

gingerbeardman, 2020-12-07:
I do understand your situation. It had exactly the same.

Good luck.

amerika13, 2020-12-07:
> I do understand your situation, I had exactly the same - you see I only use 480p output.
> 
> Good luck.

I read #117 and they reported that all resolutions had wavy lines. You stated that you had the same issue and it was resolved by changing the PSU. My issues is differing in that only one resolution appears to be affected. So, this isn't "exactly the same" and  "cause and solution have been known for a long time" does not apply here. I appreciate the assistance, but you are being dismissive without understanding why my issue is different than your own/others who have posted.

I might have an issue with my GBS 8200 or the arduino, it might be some bug in the software, it could be intererence in conjuction with another feature that is only used with the 480p preset. It might even be that all of the PSU's I tried.

Speaking of which, I just went through and tried even more PSU's.  My total is now 6. Amazon Basics branded, official Apple, official Android, official Samsung, my 2 port PSU I bought off amazon and the 2 5v USB ports on my power strip. I also changed power strips and changed USB cables and made sure I was using properly shielded official ones as well.  Nothing has changed. 480p still has waves in the output and 720p/960p seem to be crystal clear.

So, again, thank you for your time but this might be a situation where either Rama needs to comment or I can find another resource on the shmups forum.

I do plan on moving the arduino away and shielding it to see if that is causing the issue as well.

gingerbeardman, 2020-12-07:
You know about you, I know about me. Let's leave it at that.

You simply can't be helped.

amerika13, 2020-12-07:
> You know about you, I know about me. I know I had this and it affected 480p. Let's leave it at that.
> 
> You simply can't be helped.

I do not believe Rama and others appreciate you trolling.

I also read your comment that you just changed that said, "And I know you can't be helped".

Please leave this thread.

gingerbeardman, 2020-12-07:
I'm not trolling, I'm simply stating the way things are.

I think the problem is that for whatever reason you don't like the answer you've been given, because it doesn't align with your requirements, expectations, wishes. But that doesn't stop it being true. I'm sorry you're not happy but that's just how it is.

Also, I've supported this project for some time now and I believe my contributions are appreciated. Until now! 😅

amerika13, 2020-12-07:
If you don't understand the difference between the issue as I've described compared to the issue as described in #117 then please simply step away from the thread and stop being antagonistic. My issue is at a single resolution, not all resolutions like what was stated in #117 and what you confirmed your issue was as well in that thread. I have also changed out my PSU's, USB cables, power brick and even the outlet I plug into in order to eliminate your answer as being the cause of my issue. Therefor your answer does not apply.

This is the last time I reply to you as I don't want this thread dismissed/closed because of a troll.

ramapcsx2, 2020-12-07:
Calm down everyone, please.
The wavy lines can have multiple sources, only one of which is the GBS power supply.
Any other power supply, as well as the TV, etc, can also be at fault.

amerika13, 2020-12-07:
> Calm down everyone, please.
> The wavy lines can have multiple sources, only one of which is the GBS power supply.
> Any other power supply, as well as the TV, etc, can also be at fault.

My VGA to HDMI converter is active/powered. But it has also gotten the same treatment as the GBS itself in that I've changed everything that has to do with power. Maybe that adapter is crap? It's this guy specifically: https://www.amazon.com/gp/product/B07121Y1Z3/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1&fpw=alm

I have the GBS-C running into an 2019 model TCL TV and I also run it into my micomsoft interal capture card. Both show the lines so my guess is they aren't the cause. I have also tried known good HD Retrovision cables (Genesis and Saturn), Retro-Access RGB cables converted to component via a shinybow converter then into amazon basics component cables and also Monster branded Xbox component cables. All have no issues when they run through my RetroTink.

Do you have any guess why the lines would only be present at 480p @ramapcsx2 but no other resolutions?

ferolim, 2020-12-29:
Hi,

I was that opened #143. I tried different PSU but also had no luck. What solved my problem was adjusting the HTotal on the development tab. I later changed the source code to automatically apply the offset necessary to remove the wavy lines.

I'd suggest trying different HTotal values and check if it have any impact on the image.

amerika13, 2021-04-10:
BTW, I completely forgot to update this thread. But thank you @ferolim for the information. I adjusted the Htotal value up by 1 and it removed all the lines from 480p. Thank you and closing the issue.

closed 2021-04-10 by amerika13