ramapcsx2/gbs-control #165: Any chance this will work with a HD-VC9900

rfancella, 2020-12-14:
Dear sirs,
Is there any chance to get this working with a HD-VC9900.  This device appears much like a GBS device only it outputs directly to HDMI.

Leadhood, 2020-12-14:
Well I ve read that this should work, but you would need an extra adapter to inject the audio into the hdmi signal.

adgenet, 2020-12-16:
I can confirm that it works fine.
As for audio, there is no interface for it exposed on the board as far as I know, so you're limited to just video.

rfancella, 2021-01-07:
Is the instructions for this the same as with a GBS8200 including the clock generator?  The chips look like they are in the same location.  So all wires to the same pins on the same chips for both boards?

mrjackv, 2021-02-05:
Using [this](https://images-na.ssl-images-amazon.com/images/I/81KBgbR%2BtJL._SL1500_.jpg) image as reference I can tell you that:

* The pins for the installation of the esp8266 and clock generator are the same
* The capacitors cited in the hardware mod guide have the same markings and position

As for the audio input I could not find a picture where the heatsink on the HDMI chip has been removed, so I can only speculate that audio injection might be possible depending on which IC they have used. If you can send pictures of it I can have a look, still best-case scenario you will have to solder a couple of wires onto tiny legs, which might be tricky.

james194zt, 2021-08-21:
> Using [this](https://images-na.ssl-images-amazon.com/images/I/81KBgbR%2BtJL._SL1500_.jpg) image as reference I can tell you that:
> 
> * The pins for the installation of the esp8266 and clock generator are the same
> * The capacitors cited in the hardware mod guide have the same markings and position
> 
> As for the audio input I could not find a picture where the heatsink on the HDMI chip has been removed, so I can only speculate that audio injection might be possible depending on which IC they have used. If you can send pictures of it (and maybe the IC left of it) I can have a look, still best-case scenario you will have to solder a couple of wires onto tiny legs, which might be tricky.

I have just bought one and removed and scraped off the IC, the device is a MS9282 by MACRO SILICON, according to the tiny bit of information I have it is capable of audio input, but I cannot find any information on the chips pinout or a datasheet.

Anybody got any ideas? I have seen a  company selling this board with an addon board that seems to allow audio as well so it must be possible https://gamescare.com.br/produto/gbs-control-digital-gamescare-scaller-1080p-hdmi-nativo/

mrjackv, 2021-08-24:
Given that there is no available datasheet, I've looked at some reference photos [`(1)`](http://img.bj.wezhan.cn/content/sitefiles/2021049/images/9668996_VGATOHDMI%E6%8E%A5%E5%8F%A3%E5%AE%9A%E4%B9%89.jpeg) [`(2)`](https://i.imgur.com/0L0UVuy.jpg)
An ADC is needed to convert from analog audio to I2S, as shown from (2) with the ES7240 chip.
I am quite curious of the chip right below P13 on the upper left corner (according to the reference image), could you check the markings if it has any? P13 looks awfully like a 4-pin jack connector, maybe they copy-pasted the ADC on the pcb as well.
If that's not the case the only option is to add a daughter board with the i2c adc and solder some fine wires to the feet of the MS9282 chip. (If this is the case I am very skeptical that the pre-assembled board has audio)

Vitalic66, 2021-10-13:
ADC converter: 
https://digilent.com/reference/pmod/pmodi2s2/start

Desoldered in and out jacks.

[image]

Desoldered 2 pulldown resistors R66 R67 and connected 4 wires to ADC board.

red = MCLK Masterclock
white = LRCK left right channel select
orange = SCLK Serial Data clock
green = SDOUT Serial Data Out

Set the jumper in MASTER position. Tried slave first as i thought MS9282 is master, but it's other way round.

[image]

Ryoandr, 2021-12-03:
Just a heads up, I fitted my HD-VC9900 with a PCM1808 board that you can find for cheap on aliexpress, and it worked a treat. Only annoyance was the heatsink was larger than on Vitalic66 pic so I had to remove it and reattach it later with thermal double sided tape.

LawnMo, 2021-12-19:
hey,  
I was tempted to build this as it directly outputs hdmi but disregarded the fact that the VC9900 was missing a YPbPr input.  
Since both cards have similare/same chips, is it possible to wire some red/blue/green connectors similar to the yellow YPbPr connector on the 82x0 boards ?  
I'm still waiting on the oscilator and i2s2 adc converter, but I should receive the scart rgb connector soon enough to test / start modding the board, which is the main reason I got this board for, but covering YUV component would be awesome too :D

rafer22, 2021-12-19:
@LawnMo yes, you can wire like 82x0. Just need 75 ohm and 200 ohm (*3) for getting YPbPr input

LawnMo, 2021-12-19:
> @LawnMo yes, you can wire like 82x0. Just need 75 ohm and 200 ohm (*3) for getting YPbPr input

Thanks for the fast reply, I really appreciate it, do you know by any chance how/where I should hook YPbPr to ? I couldn't find good enough pics to look up where the traces go ^^ (any help appreciated as I'm not that savvy in electronics, as soon as I finish mine, I'll try to get this HD-VC9900 variant added to the wiki or write a dedicated tutorial for it, passing on the knowledge for others to build this ;) )

mafradon, 2022-01-15:
@Ryoandr 

Could you post a picture how you wired op you're pcm1808? 

I tried this mod on mine, but now I have no HDMI output at all anymore!

Ryoandr, 2022-01-15:
> @Ryoandr
> 
> Could you post a picture how you wired op you're pcm1808?
> 
> I tried this mod on mine, but now I have no HDMI output at all anymore!
[image]

this is how I tested it prior to cleaner install

rafer22, 2022-01-15:
@Ryoandr
You have to connect  3.3V to pin 2 of regulator

mafradon, 2022-01-15:
@rafer22 

So it's the middle pin of the regulator?

rafer22, 2022-01-15:
@Ryoandr yes

Ryoandr, 2022-01-15:
> @Ryoandr You have to connect 3.3V to pin 2 of regulator

no, this is the 1.8v regulator. There's 3.3v on pin 3 comming from the DC-DC buck.
Pin 2 would get me 1.8v

rafer22, 2022-01-15:
@Ryoandr you're right, I didn't see it was 1.8V 👍

LawnMo, 2022-01-15:
@Ryoandr it's hard to tell on your picture but it looks like you didn't remove R66 and R67 ? Is that optional ?

Ryoandr, 2022-01-15:
I did remove them, it's just some residual solder  that makes it look like caps still on

mafradon, 2022-01-15:
Tried to move the ground and the 3.3v supply, but no luck. 

I think I fried the HDMI output chip :(

mafradon, 2022-02-01:
New board came in today, and...
It finally works!

Only I seem to have some issues with c-sync. But the audio works fine now over HDMI.

emersonnovais, 2022-03-27:
@mafradon how do you inserted the audio? Do you have a scheme to share with us? Thanks.

RedAnon, 2022-06-22:
So what are the upsides/downsides of this VC9900 vs GBS8800? If I want just RGB SCART > HDMI (no need for sound) is this the simplest way to go?

Tgood99, 2022-08-18:
Hello all,
Been looking at this thread for a while as I am trying to gather all the parts to build a gbs-c using this board as it outputs hdmi directly. Looking to use it with a ps2 though so would need the YPbPr inputs. I know @LawnMo has already asked but I am wondering if anyone has a picture of how they wired them up. Everything else seems straightforward. Just kind of tripped up here.
Thanks!

Zronium, 2022-08-20:
Forewarning because I'm a dumdum who knows next to nothing about electronic engineering, so if anything I did is a bad idea, pls lemme know.

I'm trying to clean up connections for both the esp8266 and the 1808. I reused 2 empty spots for pin connectors, so I could make them plug and play. 

For the esp8266 I removed R45 as it only seemed to connect the 2 unused vcc pins, then I made a quick jumper from the pin on IC6 to the other end of the resistor pad (I'm out of 28awg, so I managed with 26 lol). That way I can use all 4 pins on that connector for the 8266. Only other wire needed is 5v, and that's already on its own connector.
[image]
 
 
For the 1808, I was I'm not a big fan of wires dangling from resistor pads, so I wanted something a little more substantial. I made use of the unused 4 pin connector spot on the top left of the board. The first pin is 3v3, third is GND. 2 and 4 are from the nearby IC, and there was no easy way to isolate those pins, short of cutting traces. I took a 6 pin connector, pried out pins 2, 4, 5, and 6. replaced them with right angle pins that I bend inwards a little bit, pressed them in with my iron, so they would be a little more flush. I taped off pins 2 and 4 on the board, so they wouldn't bridge, then soldered in 1 and 3. It's still not 100 sturdy, but infinitely better than it was. 
[image]

I'm planning on making a PCB design that I'll mount above with standoffs. I want it to hold the 8266, 1808, Si5351 clock, YPbPr inputs, LR audio inputs, and maybe scart, just cause. I've never done any PCB design before, so I've mostly been floundering.

If I add Scart, would I need to put in a switch to change audio sources from RCA to Scart?

I know its typically better to keep digital separate from analog, so for the 1808, just keep one side away from the rest? I'd also be sharing the SDA, and SCL lines between the esp8266 and the Si5351 clock, I have no idea what's digital and what's analog, also if it even matters with that. 

Also with grounding, I would assume all would be fine, but in my experience with guitars, I know ground loops are a thing to avoid, with that, should I only have 1 ground connection from the 9900 to my pcb, cause each pin connector has a ground that I can run up to the pcb. In addition, each of the chips I'll be mounting have multiple grounds, good/bad to connect all?

mafradon, 2022-08-20:
Here is a good instruction on how to add component to a vc9900 it's in Japanese put its readable trough translate.

Going to try this on my vc9900.

https://cpu.8bitsize.com/3mhz/index/04/vc9900-2.php

Miocastoor, 2022-10-10:
Hello everyone,

I was wondering if there was a noticeable improvement in picture quality between the VC9900 and the classic GBS boards. I’m new to all of this and I couldn’t find a lot of information about the VC9900 online.

I would like to use this scaler to plug a GameCube (SCART) to a projector (1080p) that has both VGA and HDMI inputs.

Thanks!

LawnMo, 2022-10-11:
@Miocastoor litteraly the same chips, only this has the vga->hdmi converter integrated instead of an additional extension at the back of a "regular" gbs-c.

417Arcades, 2023-05-09:
> Just a heads up, I fitted my HD-VC9900 with a PCM1808 board that you can find for cheap on aliexpress, and it worked a treat. Only annoyance was the heatsink was larger than on Vitalic66 pic so I had to remove it and reattach it later with thermal double sided tape.

Question for you; I am running same set up, Hooked up like you did. I couldn't see that well through the acrylic in the photo and removed the resistor at R65. Would this cause no sound? Also, I noticed that you had your audio inputs to the L and R input but no audio ground. I am running mono sound from a JROK pcb and Hooked up to R in and GND accordingly. Maybe this is my issue? Another thing maybe worth noting is that I get 4.94v at the main dc in but after the D2 resistor I get 4.51, is this not enough to power the 1808?
Thanks,
Mike @ 417Arcades

[image]

417Arcades, 2023-05-09:
I figured it out. Here is a cleaner write up for the integration of audio using the pcm1808 with the vc9900.

1808 - VC9900 - Position (hdmi port towards you)
BCK - R67 (remove) - right
OUT - R65 (leave) - right
LRC - R66 (remove) - right
SCK - 4th pin down on left side of (U1)
GND - unused
3.3V - bridge this to other side of 1808 (MD1)
FMY - This attaches to (-)pin off capacitor closest to the GND text on the 1808 pcb
MD1 - (bridged from 3.3V)
MDO - Unused
GND - C11 - Left
3.3V - Right leg on lower side of (IC5)
5V - Pull from top side of D2 just before C1 capacitor
Remember the direction orientation is that the DC in and HDMI out are facing you. So bottom is towards you. Then your right is right ect.

PalmCMVS, 2023-07-28:
Hello everyone. 
I have recently purchased an HD-VC9900 and a Pmodi2s2 to add audio on the HDMI output. My question is: just connecting the four cables is enough or is it also necessary to feed the Pmodi2s2 with 3.3v?

SanguineBrah, 2025-03-22:
I recently purchased a VC9900 from Ali. It is marked HD-VC9900-V1.3 and it looks like it renders these mods obsolete. Not only does it include a 4 pin JST-PH header for component video, but it also features a CJC5340 ADC with test pads marked "AL", "AR" and "GND" right next to it.

SanguineBrah, 2025-03-26:
Here are some photos of the v1.3 board:

[image]

[image]

[image]

The Ali listing I bought it from had the old version shown in the photos so you may need to contact the seller to verify what they are selling.

I will test the audio and component inputs sometime over the next couple of weeks when some parts have arrived and I will provide an update.

PalmCMVS, 2025-03-27:
> Here are some photos of the v1.3 board:
>
> [image]
>
> [image]
>
> [image]
>
> The Ali listing I bought it from had the old version shown in the photos so you may need to contact the seller to verify what they [...]

Thanks so much for the information. Could you post the link to where you bought it? I've been looking on Ali, and they only show version 1.1 in the photos.

SanguineBrah, 2025-03-27:
> Thanks so much for the information. Could you post the link to where you bought it? I've been looking on Ali, and they only show version 1.1 in the photos.

Sure! [This](https://vi.aliexpress.com/item/1005005533909209.html) is the listing I bought it from.

PalmCMVS, 2025-03-28:
> > Thanks so much for the information. Could you post the link to where you bought it? I've been looking on Ali, and they only show version 1.1 in the photos.
> 
> Sure! [This](https://vi.aliexpress.com/item/1005005533909209.html) is the listing I bought it from.

Thank you very much. I actually already purchased the HD-VC9900 PCB and the Pmodl2S2. However, due to lack of time, I haven't been able to put it together. However, this new version of the HD-VC9900, if it has good sound quality, is a much better option.

PalmCMVS, 2025-04-01:
> > Thanks so much for the information. Could you post the link to where you bought it? I've been looking on Ali, and they only show version 1.1 in the photos.
> 
> Sure! [This](https://vi.aliexpress.com/item/1005005533909209.html) is the listing I bought it from.

I messaged the seller at this link to check if they had version 1.3 available, but they only replied that they were busy and needed time to check. I haven't seen this PCB version specified anywhere, and all the photos only show version 1.1.

SanguineBrah, 2025-04-03:
I can confirm that audio injection using the onboard ADC works just fine. I wired the audio input pins from my SCART connector to the test pads and everything works as expected with no obvious quality issues.

Chandler-Kluser, 2025-11-29:
There is a board revision with GND, AR (Audio Right) and AL (Audio Left) pads near HDMI encoder as picture below:

[image]

If you have any of this board revision, there is no need to buy an ADC board, it worked like a charm here