ramapcsx2/gbs-control #167: 32X signal unstable

soniccd123, 2020-12-24:
**Setup**: 32X hooked to the GBS through a proper SCART cable. GBS hooked to CRT VGA Monitor that works perfectly with all my other devices.
**Issue**: While it shows images correctly, when there is a screen transition or some thing that affects the whole scene, the sync simply drops as if it wasn't connected.
**Tests**: 32X hooked through same cable to RGB CRT TV works flawlessly; Mega Drive hooked through same cable directly to GBS works aswell; Mega Drive game on the 32X result in the same issues.

**More Details**:
_Cable_: Sync is connected to the 32X comp. video (pin 4 on the DIN connector).

_GBS-Control settings_: Turning off SyncWatcher in the debug settings seens to stabilize the signal, but the first few lines on the screen are distorted and warped to the right.

_GBS-Control Debug info_: With SyncWatcher on these messages keep apearing at the terminal of the gbscontrol.local interface: (debug) ign. length: 0x74; 0x72, 0x5D, 0x5C are common values also shown. These messages normally appear during signal drops.

_Print Infos information_:
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0500 m:0 ht:2344 vt: 267 hpw: 208 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2345 vt: 267 hpw:  99 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2345 vt: 267 hpw: 208 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2345 vt: 267 hpw: 208 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2344 vt: 267 hpw: 208 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:058c m:0 ht:2345 vt: 267 hpw: 207 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0580 m:0 ht:2345 vt: 267 hpw: 207 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2345 vt: 266 hpw: 207 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:5 A:ffffff S:07.00.00 H-   I:00 D:0580 m:0 ht:2345 vt: 266 hpw:   1 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:4 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2345 vt: 266 hpw: 207 u:  0 s:11 S: 4 W:31
h: 429 v: 487 PLL:5 A:ffffff S:07.00.00 H-   I:00 D:0584 m:0 ht:2345 vt: 267 hpw: 145 u:  0 s:11 S: 4 W:31

_Video timmings_:
HT / scale   : 2704 512
HS ST/SP     : 8 160
HB ST/SP(d)  : 2594 612
HB ST/SP     : 2512 480
VT / scale   : 1000 512
VS ST/SP     : 1 4
VB ST/SP(d)  : 1000 38
VB ST/SP     : 24 26
IF VB ST/SP  : 6 8
CsVT         : 267
CsVS_ST/SP   : 4 1

Reading through other Issues, there is a 2018 issue that may have to do with this situation: https://github.com/ramapcsx2/gbs-control/issues/8

Hope this helps to correct this problem

soniccd123, 2020-12-31:
I've made some tests, I'm not really an expert and doesn't really understand a lot about how the signals are treated, but after reading about the resistance of the sync signal in some cables, I tried adding a potentiometer to the composite sync signal between the 32X and the GBS sync pin. After moving the pot a bit I found a spot where the signal got really stable. Upon checking the pot resistance with a ohmmeter, it read around 1.8K resistance. Tried soldering a 1K resistor in series with the sync signal of my cable and the signal is now extremelly stable, It also works on my TV in this configuration.

I don't know what has really happened and what is causing this issue. Also don't know if this is a safe fix for the problem, but wrote it down right here so it is registered for others to see. If someone who understands this more througly can give a light on what is happening and if this is adequate, please tell me.

JAOT, 2023-01-19:
Adding to the above solution, I had a SCART cable that had resistors and capacitor on the RGB lines. Adding a 75k ohm resistor and 220uf capacitor in serial to the composite sync line has drastically improved the signal.
It is fixed, clean and made a Chinese dubiously quality cable actually decent.

[image]

Like in this example from Tim Worthington, but for some reason, there's no resistors and capacitors in the composite sync....

Worth a shot, I'd say. Worked great for me.

jonnypanic, 2025-04-08:
Hi all,

I have this exact issue. Turning on Syncwatcher and disabling at the right time stablises the image. But I'd rather not have to do that every time. 

Will getting a better cable with C-Sync will fix this issue? Would just like to make sure before I spend 8000 yen on a good cable. I don't yet have the skills or knowhow to do the above cable mod.