ramapcsx2/gbs-control #168: LK7600 Clone Boards

RikuKawai, 2021-01-12:
Hi, I've built this with an LK7600 which is a clone of a GBS board and uses the same hardware, but it lacks the program jumper of the original GBS8200. What exactly do I need to ground to disable the original MCU?

goutsune, 2021-01-28:
I'm not sure how reset works on the GBS board, but I assume you can short pin 19 to pin 18 through a 1K resistor to put the MCU chip (the right one) into reset forever (or it might be pin 19 and 20 which is ground, one of these is ought to work). Or, you can desolder the chip altogether, it is not required if you install ESP8266.
[image]

ramapcsx2, 2021-01-28:
That should be right, yep.
Where did you find this board by the way? It looks nicely designed!

RikuKawai, 2021-01-28:
I just bought it as a generic Component to VGA box, monoprice sold a version of it
[image]

Doesn't seem to be available anymore, and the picture quality isn't very good, hopefully it's better if I can get this working, I had already removed the MCU but did a sloppy job with a heatgun and need to replace some capacitors I melted.

gingerbeardman, 2021-01-29:
very expensive compared to GBS!

ramapcsx2, 2021-01-29:
And unavailable, it looks like.
That's too bad really, I was looking for differently designed "carrier boards" :p

goutsune, 2021-01-29:
I have noticed XVGA box on chinese shops with "GBS-8219" on the subtitle, has anyone opened on up? Although its far more expensive than the usual gbs board.
[image]

gingerbeardman, 2021-01-29:
Looks like this inside: https://m.diytrade.com/mobi/pd/9172604/CGA_EGA_MDA_RGB_Video_TTL_Composite_to_xVGA_Converter.html

invaderlex, 2021-02-03:
Reporting here that I tried shorting the pins and then removing the chip as recomended by goutsune, it does not work. GBS NOT RESPONDING message.

goutsune, 2021-02-03:
> Reporting here that I tried shorting the pins and then removing the chip as recommended by goutsune, it does not work. GBS NOT RESPONDING message.

Can you post a photo of you wiring? After removing chip I soldered directly to remaining pads for power supply and i2c, looks like there is missing i2c connection now.

invaderlex, 2021-02-04:
> > Reporting here that I tried shorting the pins and then removing the chip as recommended by goutsune, it does not work. GBS NOT RESPONDING message.
> 
> Can you post a photo of you wiring? After removing chip I soldered directly to remaining pads for power supply and i2c, looks like there is missing i2c connection now. Also you either short pins, or remove chip, not both. But did board stop responding after you shorted pins?

I tried shorting the 2 pins no resistor. Then tried connecting the 2 pins with a 1k resistor. Both options, 19 with 20 and 19 with 18. Then removed the 1k resistor. And removed completely the mtv chip, I checked for shorts in all the pads. I resolder pin D6, GBS BOARD NOT RESPONDING in all cases. I also tried connecting D1 and D2 to pads 15 and 16 labeled in the data sheet as HSCL and HSDA. No luck. I removed the wiring and put the ESP back into my working GBS boards, I own both I was just curious if the monoprice board could work. Any new ideas?

goutsune, 2021-02-04:
Shorting pin 18 to anything without a resistor is a big no-no, that can fry any logic inputs and go further I hope it didn't affect TVia chip somehow. :(

@ramapcsx2 do you happen to know if there is a way to check if chip is alive besides getting ID with i2c?

ramapcsx2, 2021-02-04:
What you want to do with an unknown circuit like this, is removing the control CPU completely (hot air), then probing where the TiVo scaler IC I2C pins go. Use the datasheet of the TiVo to find out which pins they are (SDA, SCL).
On a GBS stock board, the pins are on the lower left corner of the IC, towards the VGA input side.

invaderlex, 2021-02-05:
> What you want to do with an unknown circuit like this, is removing the control CPU completely (hot air), then probing where the TiVo scaler IC I2C pins go. Use the datasheet of the TiVo to find out [...]

 the points [SCL GND 5V SDA] where RikuKawai connected the ESP D1 and D2, the SCL and SDA are directly connected to pins 41 and 42 of the TrueView 5725 labeled in the datasheet as 41-SCLCK and 42-SCLDA, Directly, where in the GBS8200 there are 30ohm when you measure those same points. Maybe something is toast at this point when I shorted the MTV reset pin.

ramapcsx2, 2021-02-05:
Remember that the I2C bus usually has strong pull up resistors on both signal lines. That could confuse your readings.
And if those are missing, they have to be added.