ramapcsx2/gbs-control #173: GBS with Amstrad

NZtechguy, 2021-01-19:
Hi. I left a question on the Shmups forum but it didn't seem to work.

Anyway, I have an Amstrad CPC464 and have purchased a GBS8200 for it. I'm not sure if you have had any other users with issues concerning the Amstrads. In the Amstrad wiki, they just connect the RGB and composite sync straight into the GBS and away it goes. Not for me, haha. I have the version 5.1 GBS board and it doesn't see any input from the Amstrad. If I disconnect the sync wire and tap it on the connection to the sync input of the GBS a few times, it suddenly gives me one frame with perfect colours but a shifted picture horizontally ie no sync. I have an original Amstrad green screen monitor  and if I connect that to the Amstrad CPC, I get perfect mono picture, and that uses the sync pin so I assume the sync is fine enough for that monitor. If I scope the sync line, I see what seems to be normal looking sync pulses of around 3.6 to 4V pk-pk. I also tried attenuating the sync down to 2V or 1V with a couple of series resistors and a 120ohm between sync and ground also. No joy.

Do you think I would need to clean up the syncs for the converter? What is the sync spec that the GBS expects? We use PAL here in New Zealand so I'm assuming it's auto sensing sync freq? I'm trying to get another RGB source to test the board in case I have a bad GBS.

Any help would be greatly appreciated. I'm dying to get my old Amstrad in colour :-)

Thanks,
Mark

necroware, 2021-01-19:
Hi Mark, I also messed a lot with GBS-8200 and gbs-control in the recent time, but for MDA/Hercules/CGA/EGA formats. What you describe sounds like a C-Sync issue indeed. I have three GBS-8200 in different revisions and my experience was, that GBS-8200 can work reliably only with 15kHz. Using gbs-control you can kind of go up to 21kHz, but the more you go away from 15kHz, the more it is a question of luck. Second, do you use gbs-control or plain GBS-8200? When using plain GBS-8200, it  sometimes goes mad with sync, when when you play with screen geometry (width, high, etc.). Try to play with that settings and see if something changes. If you are using gbs-control however, this is not an issue.

NZtechguy, 2021-01-19:
Thanks for your reply. I'm assuming 15.625kHz line frequency since we use PAL. I'll try to look deeper into it. The Csync is generated by a logic array chip in the computer and sent straight out. I don't think there is any way to adjust it. I haven't modified my GBS in any way yet since I'm still getting to grips with why other modders seem to just wire their Amstrads straight into RGBS wires and get a nice locked picture. I can't find anyone mentioning the settings they used for geometry so I might ask in the Amstrad forums. I haven't played with geometry (I thought it was output related) so thanks for that advice. I see an auto scan function mentioned (pressing the down button) but I see nothing happen if I press it. Should it show anything on the OSD to indicate it's scanning?

I'll try some more playing around. Also, I might be able to use the luminance side of the monitor's signals to get a sync since there is a separate luma pin I found as well. Do you know if the GBS can sync from a luma signal or does it need pure Csync signal without video signal?

necroware, 2021-01-19:
In regards of geometry here is an example: https://youtu.be/7dNEmRvxVf4
And about luma, unfortunately I don't know. As I told, I just was playing around with real CGA and such, but not with Amstrad. So the information, I can share is may be not quite applicable in your case :)

ramapcsx2, 2021-01-19:
PAL is always harder to support, both on the scaler board and your display as well (so make sure that the problem is actually with the GBS ;p).
Otherwise, go straight to the gbscontrol modifications and hope it does it :)

issalig, 2021-03-31:
I am trying to attach it to an amstrad CPC 6128 but I get "No signal" on the OSD  from GBS8200
I have a 100 ohm resistor on sync-gnd

Serial output from ESP8266 says RGB is detected but a lot of resets occur

Any hint?

```
startup
(WiFi): AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
no ext clockgen
<reset>
Chip ID: 1A 1
 G: 3E G: 3F G: 40 G: 41 G: 42

<reset>
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV 
```````````````<reset>
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV 
```````````````<reset>
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV 
```````````````<reset>
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV 
```````````````<reset>
Activity detected, input: RGB
med res

.*****
no signal
h: 511 v: 278 PLL:0 A:7f7f78 S:02.10.c2      I:00 D:aaaa m:0 ht:   0 vt: 134 hpw: 109 u: 96 s: 0 S:11 W:31
**********
<reset>
Scanning inputs for sources ...
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV 
```````````````<reset>
Activity detected, input: RGB

2345678
Format change: 2 <stable>
matched preset override > 1280x1024
ADC offset: R:3D G:42 B:3D
 (debug) ign. length: 0x1A
to1 is: 2010
 (debug) ign. length: 0x43

preset applied: 1280x1024 for PAL 50Hz 

 (debug) ign. length: 0x48

. (debug) ign. length: 0x18
* (debug) ign. length: 0x1A
** (debug) ign. length: 0x1A
* (debug) ign. length: 0x1A
* (debug) ign. length: 0x1A

no signal
h: 431 v:   2 PLL:0 A:7f7f7f S:02.1c.f2      I:00 D:0000 m:0 ht:   2 vt:   0 hpw:   0 u: 96 s: 0 S: 4 W:31
 (debug) ign. length: 0x1A
**************** (debug) ign. length: 0x46
 (debug) ign. length: 0x4C
 (debug) ign. length: 0x1A
* (debug) ign. length: 0x18
********** (debug) ign. length: 0x48
 (debug) ign. length: 0x50
 (debug) ign. length: 0x46
* (debug) ign. length: 0x52
 (debug) ign. length: 0x16

no signal
h: 431 v: 632 PLL:0 A:7f7f02 S:02.10.f2      I:00 D:0000 m:0 ht: 977 vt:   0 hpw: 314 u:384 s: 0 S:10 W:31
*********************************
no signal
h: 511 v:   2 PLL:0 A:7f7f7f S:02.10.2b      I:00 D:a6b9 m:0 ht:   0 vt:   0 hpw:   0 u:708 s: 0 S:13 W:31
** (debug) ign. length: 0x1A
* (debug) ign. length: 0x19
 (debug) ign. length: 0x1A
 (debug) ign. length: 0x1A
* (debug) ign. length: 0x1A
* (debug) ign. length: 0x14
 (debug) ign. length: 0x1A
** (debug) ign. length: 0x45
 (debug) ign. length: 0x1A
 (debug) ign. length: 0x1A
** (debug) ign. length: 0x18

<reset>
Scanning inputs for sources ...
Activity detected, input: RGB
med res

.*****
no signal
h:   0 v:   0 PLL:0 A:1c80d2 S:00.00.f2      I:00 D:aaaa m:0 ht:   0 vt:   0 hpw:   0 u: 96 s: 0 S: 4 W:31
********

```

issalig, 2021-04-02:
I have tried https://github.com/pulkomandy/gembascan and it works, not perfect but it works.
I think we could set the same parameters to gbs-control and get rid off the LM4F120 board.

NZtechguy, 2021-04-02:
That would be good. I ended up finding a SCART TV to run my Amstrad CPC464 through but I’d like to get the GBS working on pal sync at some stage. I also have a pal RGB Amiga coming which I hear is also dodgy through a GBS so I might be a candidate to do some testing. Definitely don’t want to buy a pricey LM microcontroller eval board just to set it, so GBS control would be great

issalig, 2021-04-03:
I edited the comment, I ported the code to esp8266 :) 
You can check it on https://github.com/issalig/gbs2cpc