ramapcsx2/gbs-control #174: 4x oversampling makes PS2 picture green-ish

vorvek, 2021-01-22:
I just built my GBS Control and I'm quite happy with it. I've built it mostly to record PS2 games, since the motion adaptive deinterlacing works quite well.

I have my PS2 connected through YPbPr, and found out that the default 4x oversampling makes the image very green. 1x oversampling also is slightly off, while 2x looks the best. It's saved in the profile, so it's not a big deal, but at first I thought I had broken something since my soldering skills aren't that great.

ramapcsx2, 2021-01-23:
Hey, the oversampling changes a lot in the equations, hence this is not meant to be a user setting.
I think I have the correct defaults set in all the profiles though, right?

vorvek, 2021-01-23:
It may be my unit. I was comparing the results with my CRT and and the picture, without changing anything, looked kind of green (and very, very dark). I went into the development settings and noticed that oversampling 2x made it look way closer to the image in the CRT (still dark, but I just increased the gamma in my capture device and it looked okay-ish). Also, oversampling 4x seems to create some noticeable jailbars, while 1x and 2x don't have that issue. This might be a result of the extra gamma making them way more noticeable, though. 

I'm using an OSSC as ADC for the gbs-control. I can fix the 4x oversampling colors with a bit of tweaking in the OSSC, it's not like the image is completely green, just slightly green tinted, I have to reducee the G offset in the OSSC from 126 to 118 or so for the colors to look proper.

I also run some tests with a PS1, connected through component, and I didn't need to change the colors again; the same OSSC profile still looked fine.

jordonp, 2021-01-26:
I have the same issue, I thought maybe it was just the cheap component cables I have. Turning off the "ADC auto offset calibration" setting under preferences and restarting seemed to fix it for me.  I just tried changing the oversampling setting to 1x or 2x and that also fixes it for me.

vorvek, 2021-01-27:
For clarity purposes, I took three screenshots of my PAL BIOS screen (it also happens with NTSC games):

oversampling 1x (close to what I see in my BVM)
[image]

oversampling 2x (a bit more red than 1x. In games it's not as obvious)
[image]

oversampling 4x (certainly green)
[image]

jordonp, 2021-01-27:
My console is NTSC and it looks similar to your images.

AndehX, 2021-01-27:
I've noticed the same thing on the PS2 via Component. Settings Oversampling to 2x does indeed fix the off colours.

Arnold101knight, 2021-02-03:
how you made to work pal stuff? not works for me. you used some settings in particular?

Arnold101knight, 2021-03-11:
> I didn't see this last question. PAL works out of the box for me with GBS Control.
pal ps2 and pal game? on wich tv/monitor? where you live europe or usa?

vorvek, 2021-03-11:
I live in the UK, and I use a PAL Fat PS2. I play on a BVM that only does SD (both PAL and NTSC, though), but I use the gbs control for my capture card and PAL games just work. 

I've also tested it in my computer monitor (an Acer Nitro VG270UP) and any output from the gbs control above 480p works without issues.

Arnold101knight, 2021-03-11:
> I live in the UK, and I use a PAL Fat PS2. I play on a BVM that only does SD (both PAL and NTSC, though), but I use the gbs control for my capture card and PAL games just work.
>
> I've also tested it [...]

test a standard game 480i

Arnold101knight, 2021-03-11:
also wich cable use on ps2?

vorvek, 2021-03-11:
NTSC games work fine too. Really, I'm using the gbs control for streaming and other than the color issue I raised above, it just works. 

Here I'm playing Dark Cloud NTSC: https://youtu.be/dPL0QONDoGg

And here I'm playing the PAL version instead: https://youtu.be/HMn07mAeqI4

I use HD Retrovision YPbPr cables to connect the PS2 to my Extron MAV switcher, from there I use individually shielded QFD component cables to the GBS control. The VGA output is then routed through my KVM, back to an OSSC for vga to HDMI conversion. Then it goes to my HDMI matrix, and from there, I get an output both to my monitor and to my capture card.

Of all the consoles I have, however, the PS2 easily has the worst quality picture. With the GBS control I can also see some sort of upwards light pulse that luckily isn't too noticeable in the compressed videos (although I can see it clearly in real life). This doesn't happen if I connect the PS2 directly to the OSSC out of the YPbPr switcher, and it's still there if I skip the KVM and connect the GBS Control to the OSSC directly. It's worse in PAL than it is in NTSC, but I can see it in both (it's a bit faster in NTSC).

Arnold101knight, 2021-03-11:
> NTSC games work fine too. Really, I'm using the gbs control for streaming and other than the color issue I raised above, it just works.
>
> Here I'm playing Dark Cloud NTSC: [...]

yeah but you use ossc .... -.- that change everything. pal games and pal ps2 do not work on my gbscontrol ALONE

vorvek, 2021-03-12:
The OSSC doesn't change anything. Here, I made this just for you. Connecting the VGA output directly to one of my computer screens:
https://youtu.be/wMNG-SSdNy4

Both NTSC and PAL work. If you use the 1280x1024 option for PAL it might cause your screen to recognize it as 1680x1050. It does that in my monitor; the image is 5:4 but forced to the right with some extra black space on the left.

Arnold101knight, 2021-03-12:
> The OSSC doesn't change anything. Here, I made this just for you. Connecting the VGA output directly to one of my computer screens:
> https://youtu.be/wMNG-SSdNy4
>
> Both NTSC and PAL work. If you use [...]

thanks, where you go to see that screen on gbs web ui exactly? pal mode works on a normal fhd tv? if you can try

vorvek, 2021-03-12:
That's the new WebUI, but the output console should be there, in the old WebUI as well. You access those through http://gbscontrol.local.

I'm afraid I don't have a TV, although I could see how one might be a bit too picky about what resolutions it accepts. My BVM works when I use the downscaling option, both in NTSC and PAL, for what's worth, but it accepts pretty much every SD resolution, even anamorphic stuff (although you end up with a really small image, since it's a 14" 4:3 tube).

Arnold101knight, 2021-03-12:
> That's the new WebUI, but the output console should be there, in the old WebUI as well. You access those through http://gbscontrol.local.
>
> I'm afraid I don't have a TV, although I could see how one [...]

where i get the new web ui?

vorvek, 2021-03-12:
It was recently merged to the master branch. If you programmed the ESP8266 yourself, you'll basically just have to program it again using the latest GitHub zip.

TimbleThimble, 2021-08-24:
So just curious, what was the best solution decided for this?

I definitely notice it being a tad green in 4x, but if 2x and 1x are messing with other stuff I'd rather keep 4x enabled. I have an OSSC but equally lowering the amount of green as vorvek mentions kinda crushes dark scenes and makes them even darker.

Is there any real fix to help improve colour accuracy here?

SteelBattalion, 2021-12-07:
Just posting to let others know that my experience is also the same. When using component inputs the image has a green cast over it.

vanyasem, 2023-09-14:
I have also noticed the image being green-ish in YPbPr mode on PS2. However, 480p output does not suffer from this issue. I have also tested the PS2 in RGB mode, which also does not suffer from the greenish issue (neither in 480i, nor in 480p). However, I find it inconvenient having to press the "Sync-on-green" button every time I switch from 480i to 480p in RGB mode (this makes playing games that frequently switch between the two a major pain. Gran Turismo 4, for example)

However, cycling to 2x oversampling does not improve the issue for me. All of the oversampling modes provide a greenish image for me. Disabling ADC calibration also does not do anything for me

vorvek, 2023-09-14:
From what I've seen, it's not a problem with the GBS Control, but with the PS2 output itself. Using an OSSC or a Retrotink (whichever version) will still have a green-ish tint in the image. There's this PixelFX multiconsole digital output thing that doesn't have the issue. The Youtuber EposVox mentioned it in a video, right here: https://youtu.be/SBtkpvu9hjM?t=11

I guess a more experienced person could also tinker with the analog output circuitry in the PS2 and bring the green level down, but that's way too complex for me.

ramapcsx2, 2023-09-14:
Awesome info, cheers :)

jsteel44, 2023-09-14:
> From what I've seen, it's not a problem with the GBS Control, but with the PS2 output itself. Using an OSSC or a Retrotink (whichever version) will still have a green-ish tint in the image. 

Really? I have all three devices mentioned and (using all with component) I only have the green tint with the GBS Control.

vorvek, 2023-09-14:
> Really? I have all three devices mentioned and (using all with component) I only have the green tint with the GBS Control.

You can see the Retrotink 5x Pro in the video. I can see it myself in my own Retrotink 5x Pro too.

israpps, 2023-09-14:
I only see green tint when oversampling is on 1x or 2x IIRC...

brkzlr, 2023-11-12:
I can also confirm the behaviour stated in this issue by others. I have a PAL-75002 that's been MechaPwn'd to NTSC, hooked up to GBSControl with component cables.
This issue occurs on both PAL 576i and NTSC 480i display modes.

With ADC Calibration option enabled:
- Oversampling 1x has almost no tint whatsoever, pretty much the most accurate.
- Oversampling 2x has a slight reddish tint, but it's not that noticeable.
- Oversampling 4x has that very noticeable green tint.

With ADC Calibration option disabled:
- Oversampling 1x has a red tint, more noticeable than OSR 2x with ADC on.
- Oversampling 2x has a slightly more red tint than 1x, it's as bad as the green tint on 4x with ADC on.
- Oversampling 4x has no tint, pretty much the same as OSR 1x with ADC on.

This problem does not occur on progressive display modes, tested on LaunchELF progressive video setting which is EDTV 60hz. Both OSR 1x and OSR 2x (no OSR 4x on progressive) with ADC turned off or on do not cause tint issues and have the same exact colour switching between them.
ADC off seems like it's 5% more red than ADC on but that's to be expected as it's a colour related setting. You really have to strain your eyes to see a difference though or it might be my eyes playing tricks as well.

Seeing that it's been said oversampling changes a lot and it's not meant to be an user setting, I guess the recommended option would be to turn off ADC when playing on a PS2 with interlaced display modes, but I'm not sure if that would cause colour issues or affect other stuff.