ramapcsx2/gbs-control #175: GBS board not responding!

BradfordJoseph, 2021-01-24:
Does anyone know why I would be getting “GBS board not responding!” In the readout when I go to gbscontrol.local? I doubled checked my wiring that I followed from the walkthrough on this site. Does anyone have an image of how they wired their GBS? Thanks for any help you can provide!

gingerbeardman, 2021-01-24:
Image of mine here: https://www.instagram.com/p/CCNvoRUnQMC/

Did you set Flash size when you configured your ESP board?

> Set ... Flash size to "4MB (FS:1MB OTA:~1019KB)".
> (Don't skip this step, or you won't be able to save custom configurations later.)
From: https://github.com/ramapcsx2/gbs-control/wiki/Software-Setup

BradfordJoseph, 2021-01-24:
[image]

Yeah I had it set and just re-flashed it to make sure. What are the pins you used on the esp8266? 

Someone else is having the same exact issue I am having over at this forum: 
https://shmups.system11.org/viewtopic.php?f=6&t=52172&start=3870

They said this:
I've been following this thread for a bit now and finally decided to get a gbs-8200, mod it and install gbs-control on it. I followed the wiki and bought my gbs-8200 , and used the esp12e board listed on the retrorgb site / video on gbs-control. I was able to flash the arduino module with no issues and am able to reach the gbscontrol web ui without any problem. The problem I'm having, however is no picture. Also, whenever I try to click any of the options in the gbscontrol web ui I get an error message saying "gbs board not responding!" The GBS is powering the ESP8266 with no issues and I'm pretty sure I have the SDA / SCL lines soldered in correctly ( as well as debug ) . Any ideas what the issue could be ? Thanks in advance , and thanks to Rama for all of the hard work. esp module i'm using https://www.amazon.com/gp/product/B07HF44GBT/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=retrorgb-20&creative=9325&linkCode=as2&creativeASIN=B07HF44GBT&linkId=ecf3867b2f209400ace9bfad94d4aa37
--

No responds to this person question yet. I replied that I have the same issue at that forum to see if I can get help there also.

I am very grateful for the time and effort on any help! Thanks a ton!!!

ramapcsx2, 2021-01-25:
If you've followed the instructions, haven't forgotten to set the jumper that disables the original chip, checked your power and ground wires, then all that's left trying is to swap SDA and SCL.
There's no harm in trying this.

BradfordJoseph, 2021-01-25:
Thanks @ramapcsx2 I tried swapping the SDA and SCL and got the same issue. Do you think it is because I am using a esp8266 instead of one of the other type of WiFi modules? I am very grateful for your time and help! 

[image]
[image]
[image]

BradfordJoseph, 2021-01-25:
[image]

ramapcsx2, 2021-01-26:
Check that you haven't bridged 3.3V and GND on the clock generator power cables. That looks a bit fishy.

gingerbeardman, 2021-01-26:
Not sure if this will help, but... 

I use a cheap Wemos D1 compatible ESP8266 from eBay. I can take a photo if that would help.

And I used dupont cables to connect as to this guide:
[image]

BradfordJoseph, 2021-01-26:
@ramapcsx2 I Re-soldered that power for the clock generator but still get the same issue. I am thinking about changing out the WiFi controller to see if maybe it’s acting up. It’s probably something simple I missed. I keep looking over the instruction and videos about it just to make sure. 

If I take the jumper off should the GBS 8200 act like it doesn’t have the WiFi controller or the clock? I tried that and I still would get no image from component in to a vga to hdmi out. 

Again I am very thankful for your time and looking at my issue. Sorry for all the questions!

BradfordJoseph, 2021-01-26:
[image]

I tested the clock generator and get this readout, its different from what the instructions say of “Active FrameTime Lock not necessary with external clock gen installed”. So I am guessing the clock generator is not working? 

By the way, if you can’t tell, I am a noob to this lol.

BradfordJoseph, 2021-01-26:
[image]
[image]

Here is the readout after trying some buttons in the UI. Also another picture of the board from another angle.

YoImLee, 2021-01-26:
Hi, you currently have SCL and SDA to your ESP8266 wired up the wrong way round, SCL goes to D1, SDA goes to D2. Also, looking at one of your pictures it almost seems like the SCL pad on the GBS is damaged. If you have a multimeter you could check for continuity between D1 and D2 on the ESP and SCL/SDA on the clock generator board. They share the same connections, so If everything is wired up correctly you should get a beep. Hope you figure it out soon!
[image]

BradfordJoseph, 2021-01-27:
So I switched the SCL and SDA again and applied new solder to them. I then check all the connection with a multimeter. I powered it on and it WORKED!!!!! You guys are awesome! Very thankful for your time and effort to help me with this!!!

closed 2021-01-27 by BradfordJoseph