ramapcsx2/gbs-control #176: Several scrolling tear lines with clock generator installed

YoImLee, 2021-01-25:
I've been using GBS-C for several months now and have always had this odd behavior where after playing for a while (0-30 mins or so) several tear lines appear on screen and slowly scroll up or down until they are off screen, often reappearing after some time. This is with the clock generator installed and working (according to the log message saying it's installed and its overall behavior, no more single fast scrolling tear lines etc.) 

I have 2 boards, a GBS 8200 V4.0 and an HD9800 V5.0 (more or less identical), each with their own ESP8266 and clock generator, both have their pots removed and a 100 Ohm resistor to ground, one has all the extra bypass capacitors and C11 replaced. Both boards show the same behavior.

I have seen others mention a single fixed line on screen after loading a preset, however I don't think that is what I'm experiencing here. I have tried several things to see if it would fix the problem with no results.

**Things I've tried:**
- Removing the extra bypass caps & C11 replacement
- Moving the ESP module closer to the board (directly on top of the MCU)
- Different power supplies ranging from 5-12V
- VGA out directly to a monitor
- 2 different VGA to HDMI converters
- Adding a 1nF capacitor between CLK0 on the generator and pin 40 on the scaler
- New ESP module, clock generator & GBS board
- Changing RAM speeds in the Development tab
- All presets, 480p is the only resolution that hasn't shown any tearing
- PS1, PS2, N64, SNES, all via SCART
- Sync Stripper

[Here is a video showing the tearing](https://www.youtube.com/watch?v=ANQLy373Ebo&feature=youtu.be) (1CHIP SNES via SCART, 1080p preset)

And here a picture of one of the installs with the 1nF cap still in place, keeping it for now as it doesn't seem to matter:
[image]

I'm really loving my GBS thanks to this CFW. It's so close to perfect but the tearing is so distracting, I appreciate any suggestions. Thanks

ramapcsx2, 2021-01-25:
The static tear line is a bug where the clock gen synchronizes the timings at precisely the wrong moment, so instead of being off-screen, the tear line in in the picture.
A preset change / restart should fix it, and it should not reappear after a while either.
That it does reappear for you is very odd. I don't have an explanation now :/

YoImLee, 2021-01-25:
Hi Rama, thanks for the reply. The tearing I am experiencing however seems to be different from the static line that can occur sometimes immediately after loading a preset. It consists of several lines slowly scrolling up or down across the screen several minutes after loading a preset or turning on the device, despite the clock generator seemingly doing it's job according to the log. It also happens on both my GBS boards, each with their own clock gen and ESP module. I simply connect my systems through a female SCART connector wired up to the header on the GBS so I doubt that's causing any issues.

gingerbeardman, 2021-01-25:
Can you take a video?

YoImLee, 2021-01-25:
Hi, [here's a video showing the issue.](https://www.youtube.com/watch?v=ANQLy373Ebo&feature=youtu.be) This is after letting the device run for a little while, you can see the lines under the SMW logo at the start of the video and see them slowly make their way down the screen as it plays.

gingerbeardman, 2021-01-25:
Interesting. I see on the SMW one that they're only visible them the screen is scrolling, and disappear when it isn't. Is this usual for tearing?

YoImLee, 2021-01-26:
So I may have figured it out. I decided to try a power bank and didn't get any tearing for quite some time. I then took a long extension cord and plugged in a power supply to the opposite wall in the room and again, no tearing so far. It seems a device plugged into the same outlet through a powerstrip is interfering to the point where it causes this tearing on the GBS. Makes sense I suppose since it happens on both of my units.

Now I need to figure out what device makes this happen!

AndehX, 2021-01-27:
Might explain why i've never seen this issue, as my GBS (and a bunch of other stuff) is being powered by a Kramer VA-102P5.  I got it relatively cheap on ebay, and it's honestly one of my best investments as I have quite a lot of stuff being powered via 5v (GBS, VGA to HDMI converter, HDMI switch/splitter etc)

YoImLee, 2021-01-28:
Decided to try the component video inputs on the GBS for the first time today, using my Wii the same thing happens at both 480i and 480p. Also tried every power supply I have that would work at this point and it did not make a difference so I'm officially out of ideas for now. Again, tested on both units and VGA as well as VGA to HDMI.

gingerbeardman, 2021-01-29:
Can you test with another display or removing some of the other things in your display chain?

YoImLee, 2021-01-30:
I've tested 3 displays so far, an old VGA equipped LCD monitor, a more modern 144Hz LG monitor via VGA to HDMI, and a 2015 40 inch Sony LCD TV, again via the VGA converter. The behavior is the same on all 3, as well as my capture device. The display chain I've used for most of my testing has been nothing but console via SCART or component > gbs > VGA or HDMI to display. 

I'll try the 60 inch LG TV in the living room tomorrow but I'm not holding my breath lol... I'll update soon!

gingerbeardman, 2021-01-30:
I'm happy to do some testing if that helps.

- I use 640x480 EDTV LCD
- I do not have the clock generator mod installed
- I have PS2, DC, Wii, 360.

Let me know.

YoImLee, 2021-01-31:
So I just tried the living room TV and sure enough after some time... Slowly entering the screen from the top and scrolling down
[image]

I think I'll try disabling the clock generator tomorrow. I've only used GBS-C without it very briefly and did notice some quick tearing here and there but haven't tried it long enough to see if the very slow tear lines happen also. I really appreciate the offer @gingerbeardman though I'm not sure what there is for you to test exactly, I don't think my setup is anything special apart from having the clock generator added.

TheOpponent, 2021-02-05:
I may have gotten something similar with my own clock generator. My GBS-8200 has all the recommended mods installed. I'm using PS2 with 480i component output into the GBS-8200 using the 15kHz Scale Down preset. The VGA output goes to an Extron RGB 192 to combine sync and adapt the VGA to BNC, while also splitting the VGA input to a capture box. This strong diagonal tearing appears on both my CRT and capture. It stops after power cycling the GBS-8200.
https://streamable.com/zgzz5e

gingerbeardman, 2021-02-05:
@TheOpponent thanks for your info. What game is that?

closed 2023-03-03 by YoImLee