ramapcsx2/gbs-control #177: Question: Why use of regulated 3.3V PSU is not recommended?

lovrenco, 2021-01-26:
Seems like more logical choice to use 3.3v, if you use higher voltage input to power up GBS board than 5v (which board supports)?
Can you please explain why?

ramapcsx2, 2021-01-26:
If you mean for the main power supply into the GBS, then nothing below 5V is even going to work right.
The boards use regulators to downconvert the input voltage to 3.3V. Those regulators fail if the input voltage isn't at a higher level than the target voltage.

lovrenco, 2021-01-26:
I mean for powering ESP board.
So if i use 2amp 5v on barrel input, why it is not recommended to power Wemos D1 with 3.3 regulated?

lovrenco, 2021-01-26:
See, it says on your Wiki Page. 
It is recommended to use 5V unregulated power source for ESP8266, but it would burn ESP8266 board if you connect higher voltage than 5V? 

[image]

ramapcsx2, 2021-01-26:
Okay, this is for several reasons. The WiFi has a quite significant power draw that could show up in the image.
Some GBS boards also require more current than others, and adding the ESP module to it would further strain the power line.

lovrenco, 2021-01-26:
Thanks. I will change to 5v then.

closed 2021-01-29 by lovrenco

raulferras, 2022-03-03:
Sorry to reopen this, but reading the docs was not clear to me either.
So it is safe to connect the "DC 5V-12V IN" into the 5v of the "Wemos D1 mini"? (Using a 5V power supply)