ramapcsx2/gbs-control #178: No 1024*768, 800*600 and 640*480 in memu?

lovrenco, 2021-01-26:
I cannot see those resolutions in default menu. I installed software today.
I wanted to use those resolutions for Amiga on older CRT VGA monitor.
Can i enable them somehow?

gingerbeardman, 2021-01-27:
640x480 choose the only 480 option you can see.

lovrenco, 2021-01-27:
Who can add those?
Firmware author or?
What is the procedure?
It would be really great for old machines, looking them on Crt is the way they were designed to be seen

gingerbeardman, 2021-01-28:
Well @ramapcsx2 project owner has added all presets in the past.

Here's the one for 720x480: https://github.com/ramapcsx2/gbs-control/blob/master/ntsc_720x480.h

As you can see it is a list of values, so it will require somebody who has read the documentation for the Tvia Trueview5725 (available online) and has the bench equipment and varied console hardware to test it thoroughly.

Ideally Rama could put together an outline of how he went about creating the existing presets?

lovrenco, 2021-01-28:
I can see it, seems like rather complex thing to do without knowing exactly what needs to be done. Lets hope @ramapcsx2 can help with this when he will have time. As I said, it would be nice to have option to use device with old CRT monitors which are dirt cheap these days and gameplay of retro games looks so much better on them.

Fredobedo, 2021-01-28:
Dear, 
I would also have liked to have this 640x480 preset. 
I even don't see a 720x480 preset?

lovrenco, 2021-01-28:
[image]

Would be great if it could work with monitor like this in 800*600 or 1024*768

lovrenco, 2021-01-28:
@ramapcsx2 I do not know if anyone mentioned but "DblPAL: High Res No Flicker" 640x512 resolution does not sync well frough GBS. 54Hz, 31kHz

I noticed because it worked once I had MV1200 external scandoubler for Amiga.

lovrenco, 2021-01-29:
[image]

I Also have problem wit IBM ThinkVision 15 inch LCD monitor. It does not outputs anything in 720*480 mode.

I get picture when apply 1280x720

LOG
preset applied: 1280x720 for PAL 50Hz 

source Hz: 50.00022 new out: 50.00022 clock: 80988656 (-11344)
Game:
source Hz: 49.92035 new out: 49.92035 clock: 80859280 (-129376)

However, it does not fill screen as it should

ramapcsx2, 2021-01-29:
Creating a new preset and making it work okay with the code is a lot of work. Maintaining it is difficult as well.
I've decided against adding more presets for that reason.
@gingerbeardman Sure, you'd have to come over and listen to me explain for 3 days. It is very involved ><

lovrenco, 2021-01-29:
> Creating a new preset and making it work okay with the code is a lot of work. Maintaining it is difficult as well.
> I've decided against adding more presets for that reason.
> @gingerbeardman Sure, you'd have to come over and listen to me explain for 3 days. It is very involved ><

@ramapcsx2 Robert, give us at least 1024*768, it is all rounder, supported by most of monitors in past era (cheap to acquire). 
I am sure many people would appreciate this. 

BTW, your time costs, I know, maybe if you state price for adding that option community will help with donations?

MaxPlap, 2021-02-08:
I'm also interested to have those resolutions for my old 15' lcd

Fredobedo, 2021-02-09:
Dear,
I understand that putting another Preset is taking time, 
Regarding 640x480@60fps (31khz), didn't you already implement it in the early phase of the project?
We, as arcade enthusiast, are interested at this resolution as we do have some arcade cabs (Naomi, etc.) with screen of 29" only supporting this resolution.

Such screen are becoming rare, therefore we need scalers such as "GBS Control" that can output this resoltion.

Maybe, you could explain how we could add ourselves another preset?

 Best regards

MaxPlap, 2021-02-17:
I think we should try to find it out ourselves looking at the trueview5725 datasheet. Maybe starting from the nearest working configuration: "pal_1280x1024.h"

lovrenco, 2021-04-11:
@ramapcsx2 Hey Robert, any small chance to add 1024*768 for older 15 inch VGA monitors?

lovrenco, 2021-04-12:
Fantastic beast this is. 1024*768 resolution would save the day. @ramapcsx2 c'mon, one more resolution to add, is that really such time consuming? 
I can say that in Amiga community this resolution would make a difference. Games made for Amiga are really best viewed on smaller screens.

lovrenco, 2021-04-13:
I tried everything, I cannot stretch 720x480 to full screen, and 1280x1024 is way too much for this monitor. 

[image]
[image]

ramapcsx2, 2021-04-13:
Yep, it really is this time consuming.
There are console commands (sent via terminal over USB / Serial) that can create new output formats.
Search this repository for them, but keep in mind that it's difficult!

lovrenco, 2021-04-15:
@ramapcsx2 Just a thought, is there a way to get resolution data from the original firmware somehow, to add it to gbs-control?

zajc3w, 2021-05-19:
It's an attempt to convert 1280x1024@50HZ to 1024x768@60Hz, most monitors I have don't support 50Hz anyway.
tested with 1 signal source, works same as 1280x1024 as far as I can tell
[pal_1280x1024.zip](https://github.com/ramapcsx2/gbs-control/files/6507482/pal_1280x1024.zip)

@ramapcsx2 ano pointers to where look for a problem below?
It's rgbhv signal, right side of the lines are cut off (in all profiles). no matter how I move it I get green garbage on the right.
I suspect input horizontal blanking misconfig.
[image]
[image]
[image]

lovrenco, 2021-12-26:
C'mon @ramapcsx2 Robert, try to add at least 1024*768 to the menu. 

Ton of 15" monitors are waiting for this....

sztojanov, 2022-06-07:
@ramapcsx2 please add 1024x768 output resolution, my old tft has a maximum with this.

lovrenco, 2022-12-31:
> It's an attempt to convert 1280x1024@50HZ to 1024x768@60Hz, most monitors I have don't support 50Hz anyway. tested with 1 signal source, works same as 1280x1024 as far as I can tell [...]

How do you install custom resolution exactly?

gingerbeardman, 2023-01-01:
You create the resolution definition and then compile it into the code.

idoregesz, 2023-01-10:
@gingerbeardman sorry, we cant create the exact resolution... please do it for us and share it if you can! :)

zajc3w, 2023-01-10:
It never worked stable enough, went with FPGA based converter by Luis Antoniosi mda/cga/ega  to vga  and it runs ever since with no hiccup, plug and play solution.

xjr358, 2024-03-22:
Hello, and thanks for this amazing project. I'm currently working on a solution to replace a gas-plasma screen in the Toshiba T5100 laptop, the display from the iPad 1 fits just perfectly by size (it's easy to buy and affordable as well), the only thing is that its resolution is 1024*768, I'm wondering if will it work with the GBS Control.  Are there any plans to add support of this resolution? Is it possible to help somehow to make it happen?
[image]

zajc3w, 2024-03-22:
GBS never did LVDS signalling, you would have to build custop fpga based solution to rtranlsate original lvds to ipad compatible. good luck getting specs for ipad screen lol.

gingerbeardman, 2024-03-22:
Using iPad screens is a common thing, and driver boards already exist. So, I would say just buy one of those there's no need to use GBA Control. The bonus is that the driver boards are so small you might be able to fit it inside the laptop. Good luck!

eg. AliExpress search: https://www.aliexpress.com/w/wholesale-ipad-driver-board.html?spm=a2g0o.home.search.0

zajc3w, 2024-03-22:
> Using iPad screens is a common thing, and driver boards already exist. So, I would say just buy one of those as they are specific to the task. There's no need to use a general solution like GB-Control.
> 
> The bonus is that the driver boards are so small you might be able to fit it inside the laptop. Good luck!
> 
> eg. AliExpress search: https://www.aliexpress.com/w/wholesale-ipad-driver-board.html

I understimated how much info gets leaked from apple factories, but none of them boards accept EGA or CGA as input...

gingerbeardman, 2024-03-22:
How can you make such a sweeping assumption? You can't say "none" with any certainty.

- we know some of them support VGA
- we don't know if those ones also support CGA or EGA

Anything else is just a guess. I'm a glass half full kind of guy.

xjr358, 2024-03-23:
> How can you make such a sweeping assumption? You can't say "none" with any certainty.
> 
> * we know some of them support VGA
> * we don't know if those ones also support CGA or EGA
> 
> Anything else is just a guess. I'm a glass half full kind of guy.

> Using iPad screens is a common thing, and driver boards already exist. So, I would say just buy one of those as they are specific to the task. There's no need to use a general solution like [...]

Thank you! Yes, that's exactly what I want to do, for now, my plan is to use the MCE adapter by Necroware + GBS-C + LVDS driver for the iPad screen. In the paper, the only thing that looks troublesome is that the GBS-C does not support 1024*768

zajc3w, 2024-03-23:
You can try and get the converter I mention couple comments above, worked
fine with 1024x768 VGA screens. But last time I saw for purchase was maybe
in 2022, didn't look since though.

mpalpha, 2025-02-22:
> [@ramapcsx2](https://github.com/ramapcsx2) Hey Robert, any small chance to add 1024*768 for older 15 inch VGA monitors?

[1024x768](https://imgur.com/a/Qo3eJct)

JDW1, 2025-05-30:
> Added [1024x768](https://imgur.com/a/Qo3eJct) to mine with a lot of other fixes.

@mpalpha 
Where is the firmware to add 1024x768?  
I clicked [your Github link](https://github.com/mpalpha/gbs-control) and keyword searched for "768" but it's nowhere to be found.

I have a 10" EYOYO IPS Display with native 1024x768 resolution. I am trying to get my Apple IIc to output the content to mostly fill the display.  Currently, I'm using a stock GBS-8200 board with no mods at all, along with Plamen's IIc-to-VGA adapter (which serves as an input to the GBS-8200).  I can select 1024x768 on my stock GBS-8200 board.  The problem though is the display content is shifted up too high.  The stock 8200's foolishly limited firmware prevents use of the Geometry Menu while in VGA mode, so I can't use the GBS to shift the picture down at all.  If I could, everything would be perfect.  But right now, the very top is chopped off.  Very frustrating!

I've been thinking about the GBSC mod where you add an ESP8266, but as was discussed earlier in this thread, it's totally and utterly insane that there is no 1024x768 support.  Insane!  However, I was pleasantly surprised to see your post at the end mention that 1024x768 has been added.  At least, you seem to have added it. Nevertheless, I can't find where it is.  

All said, I don't want to spend money converting my stock GBS-8200 into a GBSC (via EPS8266) if it won't output 1024x768 to my EYOYO display.  It would be, in that case, a DOWNGRADE to do the so-called "upgrade."

One last thing.  If you do indeed have the firmware required to add 1024x768, does that firmware ONLY work on the orange colored GBSC, or any GBSC device, including the orignal mod where you add the EPS8266 board to the GBS-8200?  

And just to let you know, as of May 2025, the two newer GBSC hardware devices I'm finding on AliExpress (newer than the orange GBSC) are these two:

GBSC PRO: https://www.aliexpress.com/item/1005008290934683.html

GBSC MINI: https://www.aliexpress.com/item/1005008053848771.html

Thanks.

zajc3w, 2025-05-30:
> [@mpalpha](https://github.com/mpalpha)
> Where is the firmware to add 1024x768?
> I clicked [your Github link](https://github.com/mpalpha/gbs-control) and keyword searched for "768" but it's nowhere to be found.

this repo was updated in 2024, and posts+photos about 1024x768 are from feb this year, looks like mpalpha forgot to push latest update to github...

JDW1, 2025-05-31:
I made a video just now further explaining why 1024x768 is needed, and what implications the GBS kits have for Apple II users:

https://youtu.be/wAnt1GH6MGg

mpalpha, 2026-02-11:
[image]

Updated the repo, forked source from the mcbazel odv-gbs-c 
Compiled with platform io.
[1024x768 source](https://github.com/mpalpha/gbs-control-odv/releases/tag/1.0.24)

warheat1990, 2026-10-21:
@JDW1 @VictorRicardoPSX if you are using GBSC-PRO, i've ported the 1024x768 resolution to my fork. Go to https://github.com/warheat1990/gbsc-pro/releases to download the compiled firmware.